cp2004/Octoprint-Upgrade-To-Py3 #2: [BUG] OctoPrint 1.4.1rc4 Upgrade Failure

jneilliii, 2020-08-01:
upgrade script fails with OctoPrint 1.4.1RC4.

```
Detected OctoPi installation
Checking version
OctoPrint version: 1.4.1rc4
Traceback (most recent call last):
  File "upgrade.py", line 138, in <module>
    OPRINT_GT_141 = oprint_version_gt_141(PATH_TO_VENV)
  File "upgrade.py", line 57, in oprint_version_gt_141
    if int(version_no[2]) > 0:
ValueError: invalid literal for int() with base 10: '1rc4'
```

cp2004, 2020-08-01:
Am aware of this, it has been on my list since 1.4.1rc started rolling out. It's because I was lazy and did a `string.split(".")` rather than use the SemVer regex.... To be honest I'm surprised it's taken this long to get reported, but I should have fixed it already. Will get on to it ahead of the re-write.

cp2004, 2020-08-01:
Should now be solved. New version number is 1.3.6, check you don't get caching issues downloading the script.

For OctoPi users, it changes nothing, for non-OctoPi it would ask for config directory that isn't actually needed on OctoPrint 1.4.1, properly fixed in my 2.0 re-write

closed 2020-08-08 by cp2004