cp2004/Octoprint-Upgrade-To-Py3 #3: [BUG] Upgrade fails in 1.4.1

AconiteAu, 2020-08-05:
**Describe the bug**
New to this so pls excuse any inproprieties.  When running upgrade I came across 
Traceback (most recent call last):
  File "upgrade.py", line 219, in <module>
    backup_target = re.search(r'(?<=Backup located at )(.*)(?=.zip)', backup_path_line).group()
AttributeError: 'NoneType' object has no attribute 'group'

**Potential fixes**
backup_target = re.search(r'(?<=Backup located at )(.*)(?=.zip)', backup_path_line).group()
fails because "Backup located at" output from backup command is not saved.
Adding the filling worked for me.
    206         if 'Backup' in output:
    207             backup_path_line = output

**Script output**
Please paste the output of the script here
```
pi@alderaan:~ $ python3 upgrade.py
OctoPrint Upgrade from Python 2 to Python 3 (v1.3.6)
This script requires an internet connection and it will disrupt any ongoing print jobs.
It will install the latest OctoPrint (1.4.0) and all plugins.
No configuration or other files will be overwritten
Testing: False
Press [enter] to continue or ctrl-c to quit

Detected OctoPi installation
Checking version
OctoPrint version: 1.4.1

Creating a backup so we can read the plugin list
Creating backup at octoprint-backup-20200805-193423.zip, please wait...

Traceback (most recent call last):
  File "upgrade.py", line 219, in <module>
    backup_target = re.search(r'(?<=Backup located at )(.*)(?=.zip)', backup_path_line).group()
AttributeError: 'NoneType' object has no attribute 'group'

```

**OctoPrint Version**
Octoprint 1.4.1

**Plugins installed & versions**
Plugins installed:
- Floating Navbar
- Python 3 Check
- Bed Visualizer
- PSU Control
- BLTouch Plugin
- Simple Emergency Stop
- Cancel Objects
- Resource Monitor
- Octohue
- Tab Order
- Themeify
- Autoscroll
- Navbar Temperature Plugin

**Additional information**
Screenshots, links to docs, etc.

cp2004, 2020-08-05:
Thank you for picking this up, your fix should work. I noticed you forked the repo, are you preparing a PR? If not I can fix this. 

I am actually in the process of re-writing the script, and this issue is fixed there, but it's not stable yet.

AconiteAu, 2020-08-05:
Hi,

Thanks for quick reply.  I didn’t (and still don’t) know what to do with the for so its probably best if you fix it.  Apologies.  I understand the concept of git repository but not necessarily how it works.

Regs,
Ken

cp2004, 2020-08-05:
No worries, should be fixed by the above commit. New version number is 1.3.7

closed 2020-08-08 by cp2004