cp2004/Octoprint-Upgrade-To-Py3 #4: Trouble when running the install script

dparker2112, 2020-08-17:
Hi, this is the message I got.

Detecting system info...
OctoPi version: 0.16.0
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...
Traceback (most recent call last):
  File "upgrade.py", line 554, in <module>
    plugin_keys = read_plugins_from_backup(backup_location)
  File "upgrade.py", line 370, in read_plugins_from_backup
    plugin_list = json.load(plugins)
  File "/usr/lib/python3.5/json/__init__.py", line 268, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'

Any help would be great!
Thanks,
David

cp2004, 2020-08-17:
Think I've spotted the problem. If you run `python3 --version`, which version are you running? There's a change in 3.6 that means it can accept bytes, could be that. Although I'm surprised it didn't come up in testing....

dparker2112, 2020-08-17:
Thanks!

I'm running Python 3.5.3

cp2004, 2020-08-17:
That's it. Let me see, I know how to fix it.

cp2004, 2020-08-17:
@dparker2112 Unfortunately, it appears that OctoPi 0.16 will not support OctoPrint under python 3, as it requires greater than 3.6. I thought incorrectly that 0.16 had Py3.6, but thanks for posting this. 

For now, I will mark it as not supported. *But*, I think there may be a way to update it to 3.6/7/8 to get around this. Apologies for this! Will be giving it a go in the morning.

dparker2112, 2020-08-17:
OK thanks, hopefully I won't have to reinstall.  My other two machines worked just fine so that was awesome, thanks for your work.

cp2004, 2020-08-20:
Pushed back to at least next week, sorry, I don't have a spare SD card to test it.

cp2004, 2020-08-28:
Found an SD card, so working on this hopefully this weekend. Will ping you here as I would like to test it before merging into the main branch.

dparker2112, 2020-08-28:
Cool, thanks!

⁣About David Parker

Sent from BlueMail ​

cp2004, 2020-08-28:
It looks like I may not be able to install Python 3.6 + on OctoPi 0.16 as they're not on the package repo. The only solution so far is building from source, and at that point it is not worth it and you may as well backup, download OctoPi 0.17 or newer and restore.

Will keep looking but it's not looking likely 😢

jneilliii, 2020-08-28:
https://storage.googleapis.com/octoprint/stable/octopi-stretch-lite-0.16.0.zip

cp2004, 2020-08-28:
@jneilliii I've got OctoPi 0.16 flashed, but can't find a way to install Python 3.6+. Seems it is not on the debian package repo.

jneilliii, 2020-08-28:
Might as well download the nightly octopi build 0.18 and just go straight to python 3 directly.

http://gnethomelinux.com/OctoPi/nightly/

jneilliii, 2020-08-28:
Did you do all the apt update/apt upgrade stuff? Wonder if you could add an updated apt repo? More than likely it's because of stretch vs buster.

cp2004, 2020-08-28:
Tried apt update, but not upgrade - 150something packages to update on my little Pi 0.... no thanks. Will look into adding a repo that's got Py 3.6 on - but it's more work than I thought!

jneilliii, 2020-08-28:
Yeah, I bet if you upgrade, or potentially do a dist-upgrade you could get around it, but it definitely seems like the easiest route would be to just create a backup, flash the nightly 0.18 build (it's stable, I've been running 0.18 for months) which already is built with python3, and then just restore your backup.

cp2004, 2020-08-28:
So I have managed to install Python 3.7 by changing the repo targets. I ran:

```
sudo sed -i 's/stretch/buster/g' /etc/apt/sources.list
sudo sed -i 's/stretch/buster/g' /etc/apt/sources.list.d/raspi.list

sudo apt update
sudo apt install python3
```
and it became Python 3.7

Wondering if that could be packaged into the script and run. Seems likely to me - as long as it isn't unstable to run with some buster packages and some stretch. Also need to decide whether I want to change the repo listing *back* to stretch afterwards, I'm leaning this way so that it doesn't create any surprises.

jneilliii, 2020-08-28:
I'd be more afraid of the python not working right for some reason without a dist upgrade.

cp2004, 2020-08-28:
Have decided to make it a link to a page with 'feel free to run these commands, but no guarantees!' And of course, as Gina said in Discord, 'don't cry if it breaks'!

cp2004, 2020-08-28:
Just tried this myself, and it breaks other python modules as well when upgrading. With quite a lot of effort it is *possible* to reinstall them but a waste of time when you can just backup and restore. Above commit mentions this in the readme.

closed 2020-09-03 by cp2004