cp2004/Octoprint-Upgrade-To-Py3 #10: FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.octoprint/data/backup/octoprint-backup-20200910-160051.zip'

jamiebegin, 2020-09-10:
**Describe the bug**
Running the upgrade script fails.

**Script output**
Please paste the output of the script in the box below:
```
root@octopi:/home/pi# python3 upgrade.py 
OctoPrint Upgrade to Py 3 (v2.0.3)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...
Traceback (most recent call last):
  File "upgrade.py", line 555, in <module>
    plugin_keys = read_plugins_from_backup(backup_location)
  File "upgrade.py", line 362, in read_plugins_from_backup
    with zipfile.ZipFile(backup_path, 'r') as zip_ref:
  File "/usr/lib/python3.7/zipfile.py", line 1204, in __init__
    self.fp = io.open(file, filemode)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.octoprint/data/backup/octoprint-backup-20200910-160051.zip'
```

**Script version**
Printed at the very top of the script

**OctoPrint Version**
OctoPrint 1.4.2

jamiebegin, 2020-09-10:
This occurs if you run the script as root. Instead, make sure you're using the unprivileged account that OctoPrint was installed under ("pi" user in my case, and likely most OctoPi users).

closed 2020-09-10 by jamiebegin

cp2004, 2020-09-10:
I'm not really sure what to do about this, since if a guide does not *say* use root, then you shouldn't. It really messes up the permissions on the files....

There may be a check implemented that it is not run as root, at some point to *stop* people doing this.

feutl, 2020-09-11:
@cp2004 you could just use `sudo/su` to switch to the `pi` user.
In my case I started using sudo, because the script failed the first time and asked for the "sudo password" and failed afterwards

And if it must run as the `pi` user perhaps this can help

```
su somebody <<'EOF'
command1 -p 'parameter with "quotes" inline'
command2 -p 'parameter with "quotes" inline'
command3 -p 'parameter with "quotes" inline'
EOF
```
https://stackoverflow.com/questions/17758235/how-to-execute-a-group-of-commands-as-another-user-in-bash