cp2004/Octoprint-Upgrade-To-Py3 #6: [Bug] Describe your bug

DrDrache, 2020-09-10:
**Describe the bug**
there is an error after detecting the system.
both in python27 files

**Script output**
Please paste the output of the script in the box below:
```

pi@octopi:~ $ curl -L https://get.octoprint.org/py3/upgrade.py --output upgrade.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   154  100   154    0     0    275      0 --:--:-- --:--:-- --:--:--   275
100 22216  100 22216    0     0  31202      0 --:--:-- --:--:-- --:--:-- 31202
pi@octopi:~ $   python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.0.3)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...
2020-09-10 13:11:47,024 - octolapse.__init__ - INFO - Release mode detected.
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 32, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 32, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'

Plugins installed
- DisplayLayerProgress Plugin
- Firmware Updater
- Detailed Progress
- PrettyGCode
- Webcam Iframe
- BetterHeaterTimeout
- RaiseCloud
- Themeify
- Resource Monitor
- Z Probe Offset
- Bed Visualizer
- PrintTimeGenius Plugin
- Simple Emergency Stop
- TP-Link Smartplug
- Navbar Temperature Plugin
- Marlin Flasher
- Change Filament Plugin
- Exclude Region
- Dashboard
- Floating Navbar
- EEPROM Marlin Editor Plugin
- Marlin Binary File Transfer: Protocol Mk II
- Octolapse
If you think there is something missing from here, please check the list of plugins in Octoprint
Press [enter] to continue or ctrl-c to quit

```

**Script version**
2.0.3

**OctoPrint Version**
OctoPrint 1.4.2

**Relevant plugins that may have caused errors**
If you're having trouble with a particular part, please include your installed plugins

**Additional information**
Screenshots, links to docs, etc.

DrDrache, 2020-09-10:
this bug doesn't affect the installation "for me"

cp2004, 2020-09-10:
As far as I understand it, this is an OctoLapse error, as I see it when running a backup (as the script does to read the plugin list) with that installed. Seems that it doesn't play nice when you only half-initialise the plugin subsystem to read it.

Doesn't affect installation since it is in the backup process, and the backup completes successfully.

@FormerLurker - is this something that you would need to address? You can reproduce by running `octoprint plugins backup:backup --exclude timelapses --exclude --uploads`

StarNamer, 2020-09-11:
I tried this and upgrade.py stopped outputting after the line referencing octoprint. I left it for 20 minutes, during which time atop showed python3 was using 100% cpu, so I interupted it with ctrl-c. If I left it for longer, would it eventually complete or would it be better to uninstall octolapse, upgrade and reinstall it?

`pi@octopi:~ $   python3 upgrade.py`
 `OctoPrint Upgrade to Py 3 (v2.0.4)`

`Hello!`
`This script will move your existing OctoPrint configuration from Python 2 to Python 3`
`This script requires an internet connection and it will disrupt any ongoing print jobs.`

`It will install the latest version of OctoPrint (1.4.2) and all plugins.`
`No configuration or other files will be overwritten`

`Press [enter] to continue or ctrl-c to quit`

`Detecting system info...`
`OctoPi version: 0.17.0`
`Getting OctoPrint version...`
`OctoPrint version: 1.4.2`
`Reading installed plugins...`
No handlers could be found` for logger "octoprint.plugin.core"`
`2020-09-11 22:22:45,232 - octolapse.__init__ - INFO - Release mode detected.`
`^CTraceback (most recent call last):`
`  File "upgrade.py", line 555, in <module>`

`    backup_location = create_backup(path_to_venv, config_dir)`
`  File "upgrade.py", line 341, in create_backup`
`Aborted!`
`    output, poll = run_sys_command(command)`
`  File "upgrade.py", line 95, in run_sys_command`
`    output_line = process.stdout.readline().decode('utf-8')`
`KeyboardInterrupt`
`Error in atexit._run_exitfuncs:`
`Traceback (most recent call last):`
`  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs`
`    func(*targs, **kargs)`
`  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown`
`    h.close()`
`  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 32, in close`
`    self._executor.shutdown(wait=True)`
`AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'`
`Error in sys.exitfunc:`
`Traceback (most recent call last):`
`  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs`
`    func(*targs, **kargs)`
`  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown`
`    h.close()`
`  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 32, in close`
`    self._executor.shutdown(wait=True)`
`AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'`

`pi@octopi:~ $ ^C`

cp2004, 2020-09-11:
Reading the plugin list is quite quick (<5secs) for me on a Pi4B.

Looks like something has gone wrong inside the script there, the only 'loop' it would get into is reading the plugins, or the backup command is constantly outputting something that locks up the script.

Try it again, see if it was a one-off, if not then we will investigate further tomorrow.

StarNamer, 2020-09-13:
Downloaded fresh copy of upgrade.py, rebooted Pi, ran script, same result.

StarNamer, 2020-09-13:
It appears to be the execution (run_sys_command) of the command "/home/pi/oprint/bin/python -m octoprint plugins backup:backup --exclude timelapse --exclude uploads". I've tried running this manually and, after the "Creating backup at..." line, nothing else happens. I'll leave it running overnight and see if it completes.

StarNamer, 2020-09-14:
A few seconds after I posted the last message, it failed. The key message is:

OSError: [Errno 40] Too many levels of symbolic links: u'/home/pi/.octoprint/scripts/gcode/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/snippets/sn/sn/snippets/snippets/snippets/snippets/snippets/sn/snippets/sn/snippets/snippets/sn/snippets/snippets/snippets/snippets/snippets/retractAndRaise'

Some time ago, I found that OctoPrint's script handler, it seems, doesn't allow a snippet to be called from another snippet. The workaround to get this to work, was to create a symbolic link 'snippets' in the 'scripts/gcode/snippets' folder which points to itself. Obviously, backup is trying to follow this!

I've restarted the upgrade. Backup still reports an error, but the list of plugins seems complete. However, several plugins report errors: OctoSlack, OctoLapse and at least one other.

StarNamer, 2020-09-14:
It seems DetailedPProgress has a syntax error under Python3 (incompatible use of tabs and spaces in indentation); I didn't notice that this actually lists as "Python >=2.7, <3". 

OctoLapse uses a non-latin1 character ( UnicodeEncodeError: 'latin-1' codec can't encode character '\u2018' in position 55: ordinal not in range(256)),

OctoSlack requires Pillow, which can't find some headers or library files, specifically 'jpeg'. The documentation for Pillow says it needs external libraries Zlib and libjpeg, but doesn't explain how to install them.

cp2004, 2020-09-14:
@StarNamer this is not the place to report issues with running plugins - report them to the plugin author. It's not my responsibility to make sure *every* plugin works, just that the script does.

cp2004, 2020-10-05:
Closing for now, since I have added a note about the error so hopefully it is self explanatory now 🙂 

[image]

closed 2020-10-05 by cp2004