cp2004/Octoprint-Upgrade-To-Py3 #7: [Bug] FileNotFoundError: [Errno 2] No such file or directory

feutl, 2020-09-10:
**Describe the bug**
Running the script, even multiple times ends with the same error.

**Script output**
Please paste the output of the script in the box below:
```
sudo python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.0.3)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...
Traceback (most recent call last):
  File "upgrade.py", line 555, in <module>
    plugin_keys = read_plugins_from_backup(backup_location)
  File "upgrade.py", line 362, in read_plugins_from_backup
    with zipfile.ZipFile(backup_path, 'r') as zip_ref:
  File "/usr/lib/python3.7/zipfile.py", line 1204, in __init__
    self.fp = io.open(file, filemode)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.octoprint/data/backup/octoprint-backup-20200910-152705.zip'

```

**Script version**
Printed at the very top of the script

**OctoPrint Version**
OctoPrint 1.4.2 Python 2.7.16 OctoPi 0.17.0

cp2004, 2020-09-10:
Hi @feutl - can you try running it *without* the sudo command - that will be why the backup cannot be made I think.

cp2004, 2020-09-10:
@feutl do NOT use the sudo command to run the script - not sure why that's there in the first place.

Issue is that OctoPrint is run as the `pi` user so in order to export the config, you must run as the `pi` user too.

Don't use `sudo` unless it tells you to, which is never here.

feutl, 2020-09-11:
You are quite fast in closing bugs without waiting for a response, anyhow please reopen it because without `sudo` it is not getting better

```
OctoPrint Upgrade to Py 3 (v2.0.3)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...

Plugins installed
- Printoid Plugin
- Bed Visualizer
- Prusa Mini ETA Plugin
- Tasmota
- Fullscreen Plugin
- PrusaSlicer Thumbnails
If you think there is something missing from here, please check the list of plugins in Octoprint
Press [enter] to continue or ctrl-c to quit

Root access is required to install python3-dev, please fill in the password prompt if shown
Updating package list...
[sudo] Passwort f▒r pi:
Installing python3-dev...
Traceback (most recent call last):
  File "upgrade.py", line 559, in <module>
    install_python3_dev(backup_location)
  File "upgrade.py", line 401, in install_python3_dev
    output, poll = run_sys_command(['sudo', 'apt-get', 'install', 'python3-dev', '-y'], sudo=True)
  File "upgrade.py", line 94, in run_sys_command
    output_line = process.stdout.readline().decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe4 in position 3: invalid continuation byte
```

Just different 👍

cp2004, 2020-09-11:
OK, so now we're on to a different issue, the previous one was closed. Something has gone wrong with parsing the output of the system command. I'm not entirely sure why that would happen, so do you have a different language - everything I've found is related to the language.

The solution, is to decode in `latin-1` but I will need to work out something else, since it works for most with `utf-8`

feutl, 2020-09-11:
The system is in "German"

```
pi@octopi:~ $ locale
LANG=de_AT@euro
LANGUAGE=
LC_CTYPE="de_AT@euro"
LC_NUMERIC="de_AT@euro"
LC_TIME="de_AT@euro"
LC_COLLATE="de_AT@euro"
LC_MONETARY="de_AT@euro"
LC_MESSAGES="de_AT@euro"
LC_PAPER="de_AT@euro"
LC_NAME="de_AT@euro"
LC_ADDRESS="de_AT@euro"
LC_TELEPHONE="de_AT@euro"
LC_MEASUREMENT="de_AT@euro"
LC_IDENTIFICATION="de_AT@euro"
LC_ALL=
pi@octopi:~ $ locale -a
C
C.UTF-8
de_AT@euro
de_AT.iso885915@euro
en_GB.utf8
POSIX
pi@octopi:~ $
```

cp2004, 2020-09-11:
I will see if I can reproduce it at some point this weekend.

david-dasilva, 2020-09-13:
Same issue with encoding on a french install : 
```
Root access is required to install python3-dev, please fill in the password prompt if shown
Updating package list...
Traceback (most recent call last):
  File "upgrade.py", line 595, in <module>
    install_python3_dev(backup_location)
  File "upgrade.py", line 422, in install_python3_dev
    output, poll = run_sys_command(['sudo', 'apt-get', 'update'], sudo=True)
  File "upgrade.py", line 105, in run_sys_command
    output_line = process.stdout.readline().decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa0 in position 7: invalid start byte
```
Here is the locale : 
```
LANG=fr_FR@euro
LANGUAGE=
LC_CTYPE="fr_FR@euro"
LC_NUMERIC="fr_FR@euro"
LC_TIME="fr_FR@euro"
LC_COLLATE="fr_FR@euro"
LC_MONETARY="fr_FR@euro"
LC_MESSAGES="fr_FR@euro"
LC_PAPER="fr_FR@euro"
LC_NAME="fr_FR@euro"
LC_ADDRESS="fr_FR@euro"
LC_TELEPHONE="fr_FR@euro"
LC_MEASUREMENT="fr_FR@euro"
LC_IDENTIFICATION="fr_FR@euro"
LC_ALL=
```

cp2004, 2020-09-13:
I will look into these locale issues this evening

cp2004, 2020-09-13:
@david-dasilva @feutl Would you be able to test my development branch for this please:
```
curl -L https://raw.githubusercontent.com/cp2004/Octoprint-Upgrade-To-Py3/devel/upgrade.py --output upgrade.py
python3 upgrade.py
```
You should see the version number `2.1.2` when you run it, along with `DEVELOPMENT BUILD` 

With this one, it opens the sub process stream as a text stream, which *might* solve this issue. I'm unable to test it at the moment as I would need to switch my machine to German/French to try it. Will hopefully do that this evening.

david-dasilva, 2020-09-13:
Unfortunately it did not solve the problem : 
```
Root access is required to install python3-dev, please fill in the password prompt if shown
Updating package list...
[sudo] Mot de passe de pi�:�
Traceback (most recent call last):
  File "upgrade.py", line 597, in <module>
    install_python3_dev(backup_location)
  File "upgrade.py", line 424, in install_python3_dev
    output, poll = run_sys_command(['sudo', 'apt-get', 'update'], sudo=True)
  File "upgrade.py", line 106, in run_sys_command
    output_line = process.stdout.readline()  # .decode('utf-8')
  File "/usr/lib/python3.7/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe9 in position 1: invalid continuation byte
```
(I've got unicode errors on the output, but it's due to my ssh I think)

cp2004, 2020-09-13:
OK it is going to take longer than I thought then. We might need to work in bytes throughout, or at least just for the apt stuff, not sure if pip is translated or not.

akwak2015, 2020-09-14:
Hi,
I've had the same Unicode Error on my german system.
After changing the standard local to "en_US.UTF-8 UTF-8" with the help of raspi-config the script worked for me.

cp2004, 2020-09-14:
Running my development OctoPi in French, was able to reproduce and with a pointer from @foosel I can confirm this is fixed on the main branch. When you download it again using the original guide (url `get.octoprint.org/py3/upgrade.py`, check for the version v2.1.3, no development warning. Ran through fine for me. Now just got to figure out how to go back to English.....

@feutl, @david-dasilva Please could you let me know that this runs OK for you.

closed 2020-09-14 by cp2004

david-dasilva, 2020-09-15:
It's working great now with a french system, thanks @cp2004 !

feutl, 2020-09-21:
Ran into a new error when trying to run your script

```
Updating package list...
Installing python3-dev...
Successfully installed python3-dev
Creating new Python 3 environment...
/usr/lib/python2.7/dist-packages/virtualenv.py:1090: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/virtualenv.py", line 2379, in <module>
    main()
  File "/usr/lib/python2.7/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python2.7/dist-packages/virtualenv.py", line 996, in create_environment
    download=download,
  File "/usr/lib/python2.7/dist-packages/virtualenv.py", line 926, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/usr/lib/python2.7/dist-packages/virtualenv.py", line 817, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /home/pi/oprint/bin/python3 - setuptools pkg_resources pip wheel failed with error code 2
ERROR: Failed to create Python 3 venv
Could not create new venv

Cleaning up...
Fatal Error: Exiting
```

feutl, 2020-09-28:
@cp2004 can you reopen this issue ?

cp2004, 2020-09-29:
Apologies @feutl for not getting back to you for this issue sooner. It is a different problem than the original, so I'm not going to reopen *this issue* but if you think it needs to be addressed, then please open a [new one](https://github.com/cp2004/Octoprint-Upgrade-To-Py3/issues/new/choose).

In regards to your problem posted above, that it not an error generated by my script, instead by the module `virtualenv` which is called using `python -m virtualenv --python=/usr/bin/python3 oprint` to create a new environment for OctoPrint to live in. I am not sure what to offer.

Try reinstalling virtualenv - maybe this will help.

Try: `python -m pip install --force-reinstall virtualenv`, although if there is no module named pip, then you will need to install it first using `sudo apt install python-pip`

feutl, 2020-10-05:
> Apologies @feutl for not getting back to you for this issue sooner. It is a different problem than the original, so I'm not going to reopen _this issue_ but if you think it needs to be addressed, [...]

Thanks, this fixed it and now the upgrade script worked flawless - 👍

commit 4fd93754 (2020-09-14): :bug: Fix #7 with unicode error handling as replace with `?`