cp2004/Octoprint-Upgrade-To-Py3 #9: [Bug] JSON error starting upgrade

ctgreybeard, 2020-09-10:
**Describe the bug**
I attemtped to upgrade my Octoprint and ran into an error. Right after the `Reading installed plugins...` it threw an exception.

**Script output**
Please paste the output of the script in the box below:
```
╰─➤  python3 upgrade.py 
OctoPrint Upgrade to Py 3 (v2.0.3)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
Please provide the path to your virtual environment and the config directory of OctoPrint
On OctoPi, this would be /home/pi/oprint and commands 'sudo service octoprint stop/start'
Path: /home/pi/.octoprint
Invalid venv path, please try again
Path: home/pi/.virtualenvs/octoprint
Invalid venv path, please try again
Path: /home/pi/.virtualenvs/octoprint
Path valid
Config directory: /home/pi/.octoprint
Config directory valid

To do the install, we need the service stop and start commands. (Leave blank if you don't have a service set up)
Stop command: systemctl stop octoprint.service
Start command: systemctl start octoprint.service
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...
Traceback (most recent call last):
  File "upgrade.py", line 555, in <module>
    plugin_keys = read_plugins_from_backup(backup_location)
  File "upgrade.py", line 371, in read_plugins_from_backup
    plugin_list = json.load(plugins)
  File "/usr/lib/python3.5/json/__init__.py", line 268, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'bytes'
```

**Script version**
v2.0.3

**OctoPrint Version**
OctoPrint 1.4.2

**Relevant plugins that may have caused errors**
N/A

**Additional information**
None

ctgreybeard, 2020-09-10:
A quick note, this may be caused by the back-level python3 (3.5.3) that I am running. I need to upgrade my Pi OS.

cp2004, 2020-09-10:
My question is, how did it manage to slip through the check of Python version!? Yes, you do need Python 3.6 or up, to run OctoPrint (or py2.7, but obviously we are trying to UPgrade)

If you can spot how it slipped through this:

https://github.com/cp2004/Octoprint-Upgrade-To-Py3/blob/7d6a9db86c5e6ec9bdd8ab6afe2a8b805c4f90f8/upgrade.py#L19-L23

Then I would be happy :) I will test it later.

pwr22, 2020-09-10:
@cp2004:

    if sys.version_info.major != 3 and sys.version_info.minor < 6: 

    if 3 != 3 and 5 < 6: 

    if False and True:

    if False:

So it never goes into the sanity check block 😛

cp2004, 2020-09-10:
@pwr22 Thank you! I think that was rushed in when I realised that OctoPrint needed 3.6+.

It needs to be `version !=3 or (v==3 and v < 6):` 

In kind of pseudo code 😜

Will fix this evening.

cp2004, 2020-09-10:
Should now be solved by the above commit - Hopefully this doesn't happen again!

pwr22, 2020-09-11:
@cp2004 np. Thanks for making this - it went well on my setup!

closed 2020-09-11 by cp2004