cp2004/Octoprint-Upgrade-To-Py3 #11: [Bug] Describe your bug

grantx2016, 2020-09-11:
**Describe the bug**
The script runs but does not complete, ending with an error message as described below. No upgrade is performed.

**Script output**
Output of the script follows

```
Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...
Traceback (most recent call last):
  File "upgrade.py", line 556, in <module>
    plugin_keys = read_plugins_from_backup(backup_location)
  File "upgrade.py", line 363, in read_plugins_from_backup
    with zipfile.ZipFile(backup_path, 'r') as zip_ref:
  File "/usr/lib/python3.7/zipfile.py", line 1204, in __init__
    self.fp = io.open(file, filemode)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.octoprint/data/backup/octoprint-backup-20200911-103600.zip'
root@octopi:~# Connection to octopi.centauri.lan closed by remote host.
Connection to octopi.centauri.lan closed.
```

**Script version**
2.0.4

**OctoPrint Version**
OctoPrint 1.4.2

**Relevant plugins that may have caused errors**
If you're having trouble with a particular part, please include your installed plugins > Nothing observed indicating the issue relates to a plugin.

**Additional information**
Screenshot attached showing full output during attempted upgrade.
[image]

feutl, 2020-09-11:
https://github.com/cp2004/Octoprint-Upgrade-To-Py3/issues/7#issuecomment-690367414
Run it without `sudo`

grantx2016, 2020-09-11:
@feutl Thanks! That worked. Upgrade complete.👍

closed 2020-09-11 by grantx2016