cp2004/Octoprint-Upgrade-To-Py3 #12: [Bug] Can't find Backup

elpopo-eng, 2020-09-11:
**Describe the bug**
The backup does not appear to be generated, so the script can't get plugin from backup.
So an error is generated.

**Script output**
Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...
Traceback (most recent call last):
  File "upgrade.py", line 556, in <module>
    plugin_keys = read_plugins_from_backup(backup_location)
  File "upgrade.py", line 363, in read_plugins_from_backup
    with zipfile.ZipFile(backup_path, 'r') as zip_ref:
  File "/usr/lib/python3.7/zipfile.py", line 1204, in __init__
    self.fp = io.open(file, filemode)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.octoprint/data/backup/octoprint-backup-20200911-075652.zip'

**Script version**
OctoPrint Upgrade to Py 3 (v2.0.4)

**OctoPrint Version**
OctoPrint 1.4.2

**Relevant plugins that may have caused errors**

**Additional information**

cad435, 2020-09-11:
I have exactly the same error, same versions...

elpopo-eng, 2020-09-11:
It's work, let see this issue : 
[https://github.com/cp2004/Octoprint-Upgrade-To-Py3/issues/10](url)

closed 2020-09-11 by elpopo-eng