cp2004/Octoprint-Upgrade-To-Py3 #13: ERROR: failed to restore backup

kubik256, 2020-09-11:
Is there manual way to restore backup?
Script failed and go_back.py too :-(

```bash
Manual install detected
Please provide the path to your virtual environment
Path: /home/pi/OctoPrint/venv.bak
Venv found

To revert the install, we need the service stop and start commands for OctoPrint
Stop command: sudo service octoprint stop
Start command: sudo service octoprint start
mv: nelze získat informace o '/home/pi/OctoPrint/venv.bak.bak': Adresář nebo soubor neexistuje
ERROR: failed to restore backup
Please try manually
Exiting
```

Now backup file looks like this:

```bash
pi@octobuntu:~/OctoPrint$ ls
venv.bakFAIL.bak
```

I hope that restore is still possible... I have to make double backup next time.

cp2004, 2020-09-11:
All the restore does is take the backup, and switch it with the Py3 one. So:

If you can identify your old python 2 environment, then it can just be renamed back to `venv` (or whatever it was before). You might have to move the Py3 one out of the way first, not sure if it was created? That'll be the issue if it failed to create a new one, it cannot switch it with the old one.

`mv venv.bak venv` should get you there (maybe `venv.backFAIL.bak`, I'm confused as to what the name is...)

cp2004, 2020-09-11:
@kubik256 - what happens if you run `python -m virtualenv --python=/usr/bin/python3`? That would be the command that's failing. You might not have the `virtualenv` module installed, if it is not OctoPi/Raspbian?

kubik256, 2020-09-11:
It is OctoPrint at NUC with Ubuntu.

Your command throws this:
```
python -m virtualenv --python=/usr/bin/python3
usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--reset-app-data] [--upgrade-embed-wheels] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-posix,venv}] [--seeder {app-data,pip}] [--no-seed]
                  [--activators comma_sep_list] [--clear] [--system-site-packages] [--symlinks | --copies] [--no-download | --download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip]
                  [--no-setuptools] [--no-wheel] [--no-periodic-update] [--symlink-app-data] [--prompt prompt] [-h]
                  dest
virtualenv: error: the following arguments are required: dest
```

cp2004, 2020-09-11:
@kubik256 ahh sorry it needs the destination on the end - as it is in the script. Something like `venv_test` (so it becomes `python -m virtualenv --python=/usr/bin/python3 venv_test`)

That seems to suggest to me that it is installed correctly, but is just a question of why the command doesn't work in the script, or maybe there is something else.

kubik256, 2020-09-11:
Done, it looks good.

```
python -m virtualenv --python=/usr/bin/python3 venv_test
created virtual environment CPython3.8.2.final.0-64 in 1343ms
  creator CPython3Posix(dest=/home/pi/venv_test, clear=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/pi/.local/share/virtualenv)
    added seed packages: pip==20.2.2, setuptools==49.6.0, wheel==0.35.1
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
```

kubik256, 2020-09-11:
```
pi@octobuntu:~/venv_test/bin$ source activate
(venv_test) pi@octobuntu:~/venv_test/bin$ python --version
Python 3.8.2
```

cp2004, 2020-09-11:
🤔 There must be something that goes wrong with it, but if that works then I don't know what's going on under-the-scenes.

kubik256, 2020-09-12:
If your script doesn't generate some error log outside then I'm also lost.
I'll try to fix some wheels (like PIP wheels, python-dev, venv and python itself) maybe APT purge and reinstall and then try again.

Many thanks for your help for now ;-)

cp2004, 2020-09-12:
It should log anything that comes through `stderr` to the terminal I believe, but that can be unreliable since some things only log to `stdout`. The 'failure' is triggered on a non-zero exit code, I might try and create a `--debug` flag that outputs *absolutely* everything at some point this weekend that might help.

cp2004, 2020-09-12:
Ok that was easy 🙂  implemented in https://github.com/cp2004/Octoprint-Upgrade-To-Py3/commit/64ac29c89bd7949992ba042890bbd2ead7d4a69b

Re-run the curl command and make sure you get `2.1.1` as the version of the script, when you run it

New `--debug` flag, so run the script using `python3 upgrade.py --debug` and you should get *everything* printed to terminal.

kubik256, 2020-09-13:
Cleaning up...
Finished! OctoPrint should be ready to go - IT WORKS!!!! Thanks!

I think, that version of your script I previously run was outdated - it was from OctoPrint blog and address was:
https://get.octoprint.org/py3/upgrade.py

So it wasn't directly from GitHub and maybe it is not fresh.
Many thanks for fast help ;-)

cp2004, 2020-09-13:
In theory that is just a redirect to here, but there may be some caching along the way. 👍

closed 2020-09-13 by cp2004