cp2004/Octoprint-Upgrade-To-Py3 #14: [Bug] OctoPrint-PrettyGCode plugin wasn't reinstalled after the migration

pwr22, 2020-09-11:
**Describe the bug**
Upgrade went fine aside from losing this one plugin. It does support Python 3 and reinstalling manually has gone fine.

**Script output**
Sorry I've lost the output, but it was the latest version yesterday.

**Script version**
Sorry I've lost the output, but it was the latest version yesterday.

**OctoPrint Version**
OctoPrint 1.4.2

**Relevant plugins that may have caused errors**
OctoPrint-PrettyGCode

The only errors in the output were the benign ones from Octolapse in #6. 

**Additional information**
https://plugins.octoprint.org/plugins/PrettyGCode/

jneilliii, 2020-09-11:
I had this same issue with a standard backup/restore operation last night outside of the upgrade script.

cp2004, 2020-09-11:
Conflict of IDs in the plugin repository - internally it is `prettygcode`, repository it is `PrettyGCode`.

I've reached out to @Kragrathea in https://github.com/Kragrathea/OctoPrint-PrettyGCode/issues/68, just to let them know.

@jneilliii is it easier if you change the ID, since you've got write access to the plugin repo?

jneilliii, 2020-09-11:
That was my mess up in reviewing the plugin so it probably would be easier to update the plugin repo.

jneilliii, 2020-09-11:
For whatever reason it seems the json file is generated from the md filename and not the provided id in the template part. I just renamed the md file now and the json data is showing all lowercase.

closed 2020-09-11 by cp2004