cp2004/Octoprint-Upgrade-To-Py3 #16: [Bug] OctoPrint-ExcludeRegionPlugin - Not working with Py3

petemagnusson, 2020-09-14:
**Describe the bug**
Enabling or redownloading the https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin pluggin.
Restart the OctoPi instance.
Boot up and the plugin is disabled. 

**Script output**
Please paste the output of the script in the box below:
```
HERE
```

**Script version**
Printed at the very top of the script

**OctoPrint Version**
OctoPrint 1.4.2

**Relevant plugins that may have caused errors**
https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin

**Additional information**
[image]

jneilliii, 2020-09-14:
This is not a bug with the script. Your plugin you are reporting is not python 3 compatible. Please open a ticket on that repo, it appears there is already communication with the author there and this PR.

https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/pull/39

cp2004, 2020-09-14:
It clearly says it is incompatible to Python 3 ("Python >=2.7, <3), there is nothing I can do.

I have reached out to the author previously, it is not my job to write and maintain every plugin, only this script.

closed 2020-09-14 by cp2004