cp2004/Octoprint-Upgrade-To-Py3 #17: After updating some plugins seemed to go away and unable to reinstall

Jese44, 2020-09-14:
Upgrade itself went fine and it wasnt such a big deal to reinstall a few (~3-4) plugins, but this Octohue https://plugins.octoprint.org/plugins/octohue/ does not want to install.... oh well.. it says it installs it, but the plugin itself is missing from settings.

Jese44, 2020-09-14:
Can't thell wether this bug is from this python upgrade or octohue itself, but this occured after upgrading to python. Someone else is having problem with installing octohue too... https://github.com/entrippy/OctoPrint-OctoHue/issues/39

cp2004, 2020-09-14:
Not my problem, sorry 🙁 

Provide an `octoprint.log` to the plugin author, it likely tells you *exactly* why it has failed to load.

closed 2020-09-14 by cp2004