cp2004/Octoprint-Upgrade-To-Py3 #18: confirm_no_root & os has no attribute geteuid

triconix5555, 2020-09-14:
I get this message at running the script.

C:\OctoPrint\venv\Scripts>python3 upgrade.py
Traceback (most recent call last):
  File "upgrade.py", line 570, in <module>
    if not confirm_no_root() and not args.iknowwhatimdoing:
  File "upgrade.py", line 177, in confirm_no_root
    euid = os.geteuid()
AttributeError: module 'os' has no attribute 'geteuid'

cp2004, 2020-09-15:
The script is not compatible with Windows installs, sorry 🙁 

Easiest way you can upgrade is to take a backup of your install (from inside OctoPrint), then create a new virtual environment under Python 3, then restore the backup.

And I now know that the linux check needs to come *before* the root check...

cp2004, 2020-09-15:
I've just checked and the blog post doesn't have any mention of incompatibility to Windows/MacOS. Will try and fix this now.

jneilliii, 2020-09-15:
good point, since the normal backup/restore also doesn't work in windows environment I don't think...

cp2004, 2020-09-15:
Wait a minute @jneilliii - have I just entered the details into the blog post wrong? I put in there that you could backup on windows.....

cp2004, 2020-09-15:
Aaah it is not..... let me take that blog post back!

closed 2020-09-18 by cp2004