cp2004/Octoprint-Upgrade-To-Py3 #19: Upgrade seems to fail on backup detection?

rocketguy, 2020-09-15:
Here's what happened:  

rocket@taz6:~ $ curl -L https://get.octoprint.org/py3/upgrade.py --output upgrade.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   154  100   154    0     0    203      0 --:--:-- --:--:-- --:--:--   203
100 23760  100 23760    0     0  20589      0  0:00:01  0:00:01 --:--:--  150k
rocket@taz6:~ $ python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.1.1)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...
Failed to read created backup & plugins list
Traceback (most recent call last):
  File "upgrade.py", line 386, in read_plugins_from_backup
    with zipfile.ZipFile(backup_path, 'r') as zip_ref:
  File "/usr/lib/python3.7/zipfile.py", line 1204, in __init__
    self.fp = io.open(file, filemode)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.octoprint/data/backup/octoprint-backup-20200914-210648.zip'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "upgrade.py", line 591, in <module>
    plugin_keys = read_plugins_from_backup(backup_location)
  File "upgrade.py", line 398, in read_plugins_from_backup
    print_c("If you ran this script using root (`sudo`), please make sure you don't, and run using `python3 upgrade.py`, as per the guide.")
TypeError: print_c() missing 1 required positional argument: 'color'
rocket@taz6:~ $

cp2004, 2020-09-15:
Don't run the script using root, it seems that in handling that exception it failed to print that properly.

 If you didn't, then needs investigation, I will try and fix that exception (stupid mistake?)

cp2004, 2020-09-15:
I've realised the issue - you're not logged in as Pi, so it cannot create a backup for the Pi user. If don't have OctoPrint installed as Pi, run the script with the `-c` flag, so you can enter where it's installed.

cp2004, 2020-09-15:
The exception should be fixed, but the above still applies to your usage.

rocketguy, 2020-09-15:
It is installed under the pi user, there's just no access to that remotely, so can I use sudo -u pi to run the script?

cp2004, 2020-09-15:
Just use `su - pi` and you should get there. Not sure if `sudo` is sensible here, since I have had lots of issues with people running as root...

cp2004, 2020-09-18:
@rocketguy did this work for you in the end?

rocketguy, 2020-09-21:
If I could remember what the password was for the pi account, I'm sure 
it would have.

I will probably just burn it all down and start fresh with a new octopi 
install.  And take better setup notes this time...

Thanks for your help though!

-Jay

cp2004, 2020-09-24:
Closing since it is not an issue with the script.

closed 2020-09-24 by cp2004