cp2004/Octoprint-Upgrade-To-Py3 #20: Python 3 Upgrade script fails

wmmc01, 2020-09-15:
I backed up my Octoprint with the backup menu item.
Flashed the new version, 0.17. Went through the rest of the setup as required for Octoprint.
Ran the update script for python 3, and it fails:
pi@octopi:~ $ curl -L https://raw.githubusercontent.com/cp2004/Octoprint-Upgrade-To-Py3/master/go_back.py --output go_back.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2804  100  2804    0     0  21082      0 --:--:-- --:--:-- --:--:-- 21242
pi@octopi:~ $ python3 go_back.py
OctoPrint upgrade to Python 3: go_back.py (v1.1)
This script will move your old installation back (Just in case!)
Only use it if you have used the upgrade script and it failed
Warning: There have been reports of this script failing, if it fails for you please report it to me as soon as possible. Thanks!
Press [enter] to continue, or ctrl-c to stop

Detected OctoPi installation
mv: cannot stat '/home/pi/oprint.bak': No such file or directory
ERROR: failed to restore backup
Please try manually
Exiting
pi@octopi:~ $

cp2004, 2020-09-15:
That's not the upgrade script! `go_back.py` is to revert back from python 3 to python 2, should something fail. It won't work, if you haven't already upgraded.

Use `upgrade.py` as suggested in the guide

wmmc01, 2020-09-16:
You are correct of course, not sure how I copied the wrong script sorry for the bother!

closed 2020-09-16 by cp2004