cp2004/Octoprint-Upgrade-To-Py3 #21: [Bug] Repository changed its "Suite"

UdoKuehnen, 2020-09-15:
Upgrade failed:

E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'
ERROR: failed to update package list

Upgrade steps:

pi@octopi:~ $ curl -L https://get.octoprint.org/py3/upgrade.py --output upgrade.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   154  100   154    0     0   1107      0 --:--:-- --:--:-- --:--:--  1107
100 23883  100 23883    0     0  43983      0 --:--:-- --:--:-- --:--:-- 96302
pi@octopi:~ $ python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.1.3)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...

Plugins installed
- DisplayLayerProgress Plugin
- Detailed Progress
- Filament Manager
- FileManager
- Cost Estimation
- Dashboard
- Sidebar Webcam
If you think there is something missing from here, please check the list of plugins in Octoprint
Press [enter] to continue or ctrl-c to quit

Root access is required to install python3-dev, please fill in the password prompt if shown
Updating package list...
[sudo] password for pi:
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'
ERROR: failed to update package list
Please try manually

Cleaning up...
Fatal error: Exiting

Another Upgrade for a second printer worked well, without problems.

Many Thaks Udo

cp2004, 2020-09-15:
Something up with your instance updating it's package list. Not much I can do about that. The command run was `sudo apt-get update` so if that refuses to run, then you need to fix that. 

You could look at this topic on the Raspberry Pi forums, for potential solutions, https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=245022

It recommends running `sudo apt update -y`, then it should work.

cp2004, 2020-09-18:
@UdoKuehnen - did you manage to get this sorted?

UdoKuehnen, 2020-09-18:
> 
> 
> @UdoKuehnen - did you manage to get this sorted?

@cp2004 - Unfortunately I had no time until now. I will try today or tomorrow, I will also inform you about the success or failure.

UdoKuehnen, 2020-09-18:
O.K. I upgraded the Suit from testing to stable version successful.
Octoprint is working well.

So, time to upgrade to python3
unfortunately NOT successful, see below.

pi@AnetA8Plus:~ $ curl -L https://get.octoprint.org/py3/upgrade.py --output upgr                                                                       ade.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   154  100   154    0     0    649      0 --:--:-- --:--:-- --:--:--   652
100 24687  100 24687    0     0  36144      0 --:--:-- --:--:-- --:--:-- 82842
pi@AnetA8Plus:~ $ python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.1.5)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Pyt                                                                       hon 3
This script requires an internet connection and it will disrupt any ongoing prin                                                                       t jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
OctoPi version: 0.17.0
Traceback (most recent call last):
  File "upgrade.py", line 598, in <module>
    path_to_venv, commands, config_dir = get_env_config(octopi_valid)
  File "upgrade.py", line 268, in get_env_config
    if venv_path and not check_venv_python(venv_path):
  File "upgrade.py", line 344, in check_venv_python
    if version_str.endswith("+"):
AttributeError: 'list' object has no attribute 'endswith'

cp2004, 2020-09-18:
Will look into it ASAP, that was a bug introduced by my latest commit.... 🤦‍♂️

Stay tuned, I'll get it turned around within the hour

cp2004, 2020-09-18:
@UdoKuehnen Should now be solved, by c8afa608f4d549a51c26d916b873d5c02dce127b

UdoKuehnen, 2020-09-18:
Wow, that's very fast.
I'll try it and will give you a feedback soon. 😄

UdoKuehnen, 2020-09-18:
Many Thanks Charlie,

Great, this was the second printer I could upgrade successful.
The thirt one should now be not a problem.
(Anet A8[raspberry P4], Anet A8Plus[raspberry P4, K8200[raspberry P3])
All of them work fine. 
Only the PlugIn "CostEstimation" is compiled against python 2, but that is not a proble. 😃 

pi@Anet-A8-Plus:~ $ curl -L https://get.octoprint.org/py3/upgrade.py --output upgrade.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   154  100   154    0     0    529      0 --:--:-- --:--:-- --:--:--   529
100 24666  100 24666    0     0  45847      0 --:--:-- --:--:-- --:--:--  133k
pi@Anet-A8-Plus:~ $ python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.1.6)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...
2020-09-18 20:25:23,932 - octolapse.__init__ - INFO - Release mode detected.
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 32, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 32, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'

Plugins installed
- DisplayLayerProgress Plugin
- Detailed Progress
- Filament Manager
- FileManager
- Change Filament Plugin
- Cost Estimation
- Dashboard
- Sidebar Webcam
- Octolapse
If you think there is something missing from here, please check the list of plugins in Octoprint
Press [enter] to continue or ctrl-c to quit

Root access is required to install python3-dev, please fill in the password prompt if shown
Updating package list...
[sudo] password for pi:
Installing python3-dev...
Successfully installed python3-dev
Creating new Python 3 environment...
/usr/lib/python2.7/dist-packages/virtualenv.py:1090: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Successfully created Python 3 environment at /home/pi/oprint

Installing OctoPrint... (This may take a while - Do not cancel!)
Collecting required packages
Installing collected packages
OctoPrint successfully installed!

Downloading OctoPrint's plugin repo

Installing Detailed Progress
Collecting required packages
Installing collected packages
Plugin Detailed Progress successfully installed
Installing FileManager
Collecting required packages
Installing collected packages
Plugin FileManager successfully installed
Installing CostEstimation
Collecting required packages
Installing collected packages
Plugin CostEstimation successfully installed
Installing FilamentManager
Collecting required packages
Installing collected packages
Plugin FilamentManager successfully installed
Installing Octolapse
Collecting required packages
Installing collected packages
Plugin Octolapse successfully installed
Installing DisplayLayerProgress
Collecting required packages
Installing collected packages
Plugin DisplayLayerProgress successfully installed
Installing Change_Filament
Collecting required packages
Installing collected packages
Plugin Change_Filament successfully installed
Installing OctoPrint-Webcamsb
Collecting required packages
Installing collected packages
Plugin OctoPrint-Webcamsb successfully installed
Installing OctoPrint-Dashboard
Collecting required packages
Installing collected packages
Plugin OctoPrint-Dashboard successfully installed

Cleaning up...
Finished! OctoPrint should be ready to go
Once you have verified the install works, you can safely remove the folder /home/pi/oprint.bak
If you want to go back (If it doesn't work) to Python 2 download the file at:
https://raw.githubusercontent.com/cp2004/Octoprint-Upgrade-To-Py3/master/go_back.py

[image]

cp2004, 2020-09-18:
Excellent, enjoy!

closed 2020-09-18 by cp2004