cp2004/Octoprint-Upgrade-To-Py3 #22: [Bug] When I run the script, it says that I'm already at Python 3, but I'm not

simpat1zq, 2020-09-18:
OP 1.4.2. Latest version of the script. 2.1.6

When I run the script, I get this error: "Virtual environment is already Python 3, are you sure you need an upgrade?"

Charlie figured this part out, I'm just summarizing the discussion from the forum. 
The reason for this is that the venv/bin/python exectuable version outputs as 2.7.15+, with the '+' being the problem. Charlie said he changed the code to drop the '+', but I'm still getting the error. 

I don't know much Python, but to troubleshoot, I add the following debug to around line 344, which is where the version check takes place(I think).

`        print("1")
        print(line)
        if line.endswith("+"):
            line = line[:-1]
            print("2")
        print("3")
        print(line)
`
When I run that, I get the following output. This leads me to believe that the IF statement is not evaluating to true, even though it seems like it obviously should evaluate to true. On a hunch, I thought maybe there is whitespace at the end of the string, and sure enough, when I added a line = line.strip() right before the IF statement, it started working properly. Please take a look and make sure that isn't something crazy that I did. I'll probably wait to hear back from you before proceeding. 

`Path: /home/hass/OctoPrint/venv
1
Python 2.7.15+

3
Python 2.7.15+

Virtual environment is already Python 3, are you sure you need an upgrade?
Please try again
Path:
`

cp2004, 2020-09-19:
Adding whitespace stripping in there should not be a problem. 

To test against the version regex, the input was `line.rstrip()`, so I can just bring that forward. When I tested it mocking your version number it didn't have any whitespace so I assumed it worked. 

I'll change that around when I get the chance later today.

simpat1zq, 2020-09-19:
Thanks. It all worked perfectly after that hurdle. 

That said, I think that how very you are grabbing the version, it's putting a new line character at the end. That's why it wasn't detecting the + at the end.  I realized it when I ran the script again and looked at the debug lines I put in. The output from before the IF had a new line after it. The output after the Trim didn’t have the new line.

cp2004, 2020-09-19:
Should now be *actually* solved, with https://github.com/cp2004/Octoprint-Upgrade-To-Py3/commit/7fead9158ac6b1cc7a8ab7e26aaf5113c9fd0241

Thank you so much for both your patience & help with debugging & solving this issue, we should be over the line now!

closed 2020-09-20 by cp2004