cp2004/Octoprint-Upgrade-To-Py3 #24: [Bug] python3 segfaults after upgrade

DemoFreak, 2020-09-21:
**Describe the bug**
After successfully running the upgrade script python3 segfaults.
Even running it as another user (root) segfaults, so I guess it has nothing to do with the environment.

**Script output**
```
pi@octopi:~ $   python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.1.7)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.4.1
Reading installed plugins...
2020-09-21 11:25:59,158 - octolapse.__init__ - INFO - Release mode detected.
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 32, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 32, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'

Plugins installed
- DisplayLayerProgress Plugin
- LayerDisplay
- Floating Navbar
- Preheat Button
- Printoid Plugin
- GcodeEditor
- Smart Preheat
- The Spaghetti Detective
- Cancel Objects
- FileManager
- Octolapse
- Bed Leveling Wizard
- PrusaSlicer Thumbnails
- Themeify
- PrintTimeGenius Plugin
- Navbar Temperature Plugin
If you think there is something missing from here, please check the list of plugins in Octoprint
Press [enter] to continue or ctrl-c to quit

Root access is required to install python3-dev, please fill in the password prompt if shown
Updating package list...
Installing python3-dev...
Successfully installed python3-dev
Creating new Python 3 environment...
/usr/lib/python2.7/dist-packages/virtualenv.py:1090: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Successfully created Python 3 environment at /home/pi/oprint

Installing OctoPrint... (This may take a while - Do not cancel!)
Collecting required packages
Installing collected packages
OctoPrint successfully installed!

Downloading OctoPrint's plugin repo

Installing Navbar Temp
Collecting required packages
Installing collected packages
Plugin Navbar Temp successfully installed
Installing FileManager
Collecting required packages
Installing collected packages
Plugin FileManager successfully installed
Installing GcodeEditor
Collecting required packages
Installing collected packages
Plugin GcodeEditor successfully installed
Installing Themeify
Collecting required packages
Installing collected packages
Plugin Themeify successfully installed
Installing Preheat Button
Collecting required packages
Installing collected packages
Plugin Preheat Button successfully installed
Installing OctoPrint-FloatingNavbar
Collecting required packages
Installing collected packages
Plugin OctoPrint-FloatingNavbar successfully installed
Installing Octolapse
Collecting required packages
Installing collected packages
Plugin Octolapse successfully installed
Installing DisplayLayerProgress
Collecting required packages
Installing collected packages
Plugin DisplayLayerProgress successfully installed
Installing OctoPrint-Cancelobject
Collecting required packages
Installing collected packages
Plugin OctoPrint-Cancelobject successfully installed
Installing OctoPrint-PrintTimeGenius
Collecting required packages
Installing collected packages
Plugin OctoPrint-PrintTimeGenius successfully installed
Installing LayerDisplay
Collecting required packages
Installing collected packages
Plugin LayerDisplay successfully installed
Installing OctoPrint-BedLevelingWizard
Collecting required packages
Installing collected packages
Plugin OctoPrint-BedLevelingWizard successfully installed
Installing Access Anywhere - The Spaghetti Detective
Collecting required packages
Installing collected packages
Plugin Access Anywhere - The Spaghetti Detective successfully installed
Installing PrusaSlicer Thumbnails
Collecting required packages
Installing collected packages
Plugin PrusaSlicer Thumbnails successfully installed
Installing OctoPrint-Printoid
Collecting required packages
Installing collected packages
Plugin OctoPrint-Printoid successfully installed
Installing Smart Preheat
Collecting required packages
Installing collected packages
Plugin Smart Preheat successfully installed

Cleaning up...
Finished! OctoPrint should be ready to go
Once you have verified the install works, you can safely remove the folder /home/pi/oprint.bak
If you want to go back (If it doesn't work) to Python 2 download the file at: 
https://raw.githubusercontent.com/cp2004/Octoprint-Upgrade-To-Py3/master/go_back.py
pi@octopi:~ $
```

```
pi@octopi:~ $ python --version
Python 2.7.16
pi@octopi:~ $ python3 --version
Segmentation fault
pi@octopi:~ $ sudo su -

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

root@octopi:~# python3
Segmentation fault
root@octopi:~# logout
```

**Script version**
OctoPrint Upgrade to Py 3 (v2.1.7)

**OctoPrint Version**
OctoPrint 1.4.1

cp2004, 2020-09-21:
Something very broken with your underlying system I think. It barely touches the system wide python install, with just one exception, everything else is installed into a new virtual environment, created from using Python 3 instead of 2.

The only thing that touches system wide installation is installing python3-dev, using apt - and it appears that ran through successfully, from your paste above. That's the only thing that *could* potentially cause this, if you have a slightly corrupted Python 3 install, or  All I can recommend is there must be some apt commands to repair broken packages, but I don't know them. 

There's not a lot else I can suggest apart from that, if you have any thoughts let me know.

DemoFreak, 2020-09-21:
The underlying system is a clean octopi 0.17 installed from image file.
The only thing I changed is adding the pi user to the sudoers.

Something happened whilst the upgrade, as the upgrade script was running fine (so python3 was working then), and immediately afterwards python3 segfaults.

I have no idea but reinstalling the system from scratch, as apt seems to depend on python3, so reinstalling python3 via apt fails too.

DemoFreak, 2020-10-03:
Seems to be a SD card issue.

Reinstalled the system from scratch now and tried the upgrade several times, each attempt failed for some other reason.
Sorry for disturbance.

closed 2020-10-03 by DemoFreak