cp2004/Octoprint-Upgrade-To-Py3 #25: [Bug] Installation failes after fresh install

nicedevil007, 2020-09-23:
I installed octopi from scratch => nightly build from today (0.18, 2020-09-23).

I get this error:

[image]

jneilliii, 2020-09-23:
The nightly builds already come bundled with Python3, you don't need to upgrade.

nicedevil007, 2020-09-23:
ahhaha... ok then it was just an old version of 0.18 that didn't come with python3 activated (my installation was a bit older since is reinstalled from scratch today)

ok thanks! problem solved, but now the bed level visualizer is lacking of support. but not a prob from you :D

closed 2020-09-23 by nicedevil007

jneilliii, 2020-09-23:
There's a known issue on the Bed Leveler's plugin Repo, it's the same thing...you just need to run `sudo apt install libatlas-base-dev` on your pi and then the plugin will work fine.

nicedevil007, 2020-09-23:
> There's a known issue on the Bed Leveler's plugin Repo, it's the same thing...you just need to run `sudo apt install libatlas-base-dev` on your pi and then the plugin will work fine.

<3