cp2004/Octoprint-Upgrade-To-Py3 #26: [Bug] upgrade.py script fails while reading OctoPrint version

shadowpanther, 2020-10-02:
**Describe the bug**
I tried the upgrade on raspbian 9, but it didn't have python3. So I upgraded to raspbian 10. Now the script runs, but fails after I enter all the paths.

**Script output**
Please paste the output of the script in the box below:
```
panther@raspistripes:~ $ python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.1.8)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
Please provide the path to your virtual environment and the config directory of OctoPrint
On OctoPi, this would be `/home/pi/oprint` (venv), `/home/pi/.octoprint` (config)
and service commands commands `sudo service octoprint stop/start`
Path: /home/panther/OctoPrint/venv
Path valid
Config directory: /home/panther/.octoprint
Config directory valid

To do the install, we need the service stop and start commands. (Leave blank if you don't have a service set up)
Stop command: service octoprint stop
Start command: service octoprint start
Getting OctoPrint version...
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 163, in _run_module_as_main
    mod_name, _Error)
  File "/usr/lib/python2.7/runpy.py", line 111, in _get_module_details
    __import__(mod_name)  # Do not catch exceptions initializing package
  File "/home/panther/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/__init__.py", line 7, in <module>
    import logging as log
  File "/usr/lib/python2.7/logging/__init__.py", line 26, in <module>
    import sys, os, time, cStringIO, traceback, warnings, weakref, collections
  File "/usr/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
Failed to find OctoPrint install
If you are not on OctoPi, please check you entered the correct path to your virtual environment
```

**Script version**
v2.1.8

**OctoPrint Version**
OctoPrint 1.4.2

**Relevant plugins that may have caused errors**
If you're having trouble with a particular part, please include your installed plugins

**Additional information**
Screenshots, links to docs, etc.

cp2004, 2020-10-02:
Does OctoPrint work properly? That's the error received when running `python -m octoprint --version` (or *technically* `/home/panther/OctoPrint/venv/bin/python -m octoprint --version`

If that doesn't work, then the script cannot work. It looks like it could be a broken install, maybe you want to try reinstalling:
1. Activate the virtual environment
2. Run `python -m pip install --force-reinstall --no-cache-dir octoprint` to reinstall it. No config etc. is overwritten, just the python packages.

shadowpanther, 2020-10-02:
OctoPrint doesn't start, and pip doesn't work too:
```
(venv)panther@raspistripes:~/OctoPrint/venv $ python -m pip install --force-reinstall --no-cache-dir octoprint
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/panther/OctoPrint/venv/lib/python2.7/site-packages/pip/__main__.py", line 16, in <module>
    from pip._internal.main import main as _main  # isort:skip # noqa
  File "/home/panther/OctoPrint/venv/local/lib/python2.7/site-packages/pip/_internal/main.py", line 9, in <module>
    import logging
  File "/usr/lib/python2.7/logging/__init__.py", line 26, in <module>
    import sys, os, time, cStringIO, traceback, warnings, weakref, collections
  File "/usr/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
```

cp2004, 2020-10-02:
> OctoPrint only stopped running after I upgraded raspbian from 9 to 10
This is exactly why I don't recommend it, and to install from scratch. It just doesn't play nice with Python virtual environments, and installed modules. If you've figured it out, this issue can be closed then. If Python works, the script should work too.

shadowpanther, 2020-10-02:
The script also required me to install `python3-requests`, without it it only upgraded the OctoPrint itself, not the plugins. Everything upgraded now, thanks for pointing me in the right direction.

closed 2020-10-02 by shadowpanther

cp2004, 2020-10-02:
I thought I added a check for that closer to the start, maybe it should 'check it is installed' and warn you if not. Maybe I'll implement that soon.