cp2004/Octoprint-Upgrade-To-Py3 #28: blocked because octopi version is too old

pjratl, 2020-10-09:
I have upgraded the system using the standard apt-get commands to latest version but your script is checking for a octopi version file I believe any way around this?

cp2004, 2020-10-09:
If you think it will work, try using the command line flag `--custom` and it should get around that check. Be warned, potential system instability lies ahead.

cp2004, 2020-10-09:
Also, standard `apt upgrade` is not sufficient.

pjratl, 2020-10-09:
yeah just figured that out manually installing building python 3.8 now

pjratl, 2020-10-09:
ok upgraded python to3.8
tried the --custom switch

got this
Detecting system info...
OctoPi version: 0.16.0
Your OctoPi install does not support upgrading OctoPrint to Python 3 - Please upgrade your install.
Details: https://github.com/cp2004/Octoprint-Upgrade-To-Py3#what-do-i-do-if-my-system-is-not-supported
Looks like your OS is not linux

pjratl, 2020-10-11:
closing this issue just did a clean install and restore everything works fine

closed 2020-10-11 by pjratl