cp2004/Octoprint-Upgrade-To-Py3 #29: Plugins are not installed?

wilbo70, 2020-10-12:
Hi. 
I performed the steps, I have Octopi Version 0.17.0, running on Raspberry Pi 4 Model B Rev 1.2, and Octoprint Version 1.4.2.
I ran into the following issue, where no plugins were installed. So I used the go_back script, which worked!

From the terminal session:
Successfully created Python 3 environment at /home/pi/oprint

Installing OctoPrint... (This may take a while - Do not cancel!)
Collecting required packages
Installing collected packages
OctoPrint successfully installed!
Required dependency requests is missing... No plugins can be installed
OctoPrint has been installed, but no plugins have

cp2004, 2020-10-12:
Try running `python3 -m pip install requests` (it should be there by default on OctoPi 0.17, but hey). If that does not work, you may have to install pip first (`sudo apt update`, then run `sudo apt install python3-pip`)

cp2004, 2020-10-16:
Hey @wilbo70 - did you get anywhere with this?

wilbo70, 2020-10-17:
> Hey @wilbo70 - did you get anywhere with this?

Hi, sorry for the wait. I didn't find the time during the week. I just tried, and I did have to install pip, it wasn't there? I then ran the script again (checked that you hadn't published a  new version first :) ). And had the same results I am afraid?
I have added the full SSH history

login as: pi
pi@192.168.2.24's password:
Linux octopi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Oct 16 15:05:50 2020 from 192.168.2.10

------------------------------------------------------------------------------
Access OctoPrint from a web browser on your network by navigating to any of:

    http://octopi.local
    http://192.168.2.24

https is also available, with a self-signed certificate.
------------------------------------------------------------------------------
This image comes without a desktop environment installed because it's not
required for running OctoPrint. If you want a desktop environment you can
install it via

    sudo /home/pi/scripts/install-desktop
------------------------------------------------------------------------------
OctoPrint version : 1.4.2
OctoPi version    : 0.17.0
------------------------------------------------------------------------------

pi@octopi:~ $ ls
go_back.py  libcoap  mjpg-streamer  OctoPrint  oprint  scripts  upgrade.py
pi@octopi:~ $ sudo python3 -m pip install requests
[sudo] password for pi:
/usr/bin/python3: No module named pip
pi@octopi:~ $ python3 -m pip install requests
/usr/bin/python3: No module named pip
pi@octopi:~ $ sudo apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [331 kB]
Fetched 13.4 MB in 6s (2,412 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
66 packages can be upgraded. Run 'apt list --upgradable' to see them.
pi@octopi:~ $ sudo apt install python3-pip
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages were automatically installed and are no longer required:
  distro-info-data iso-codes python-apt-common rpi-eeprom-images
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  python3-asn1crypto python3-cffi-backend python3-crypto python3-cryptography
  python3-dbus python3-entrypoints python3-gi python3-keyring
  python3-keyrings.alt python3-pkg-resources python3-secretstorage
  python3-setuptools python3-six python3-wheel python3-xdg
Suggested packages:
  python-crypto-doc python-cryptography-doc python3-cryptography-vectors
  python-dbus-doc python3-dbus-dbg gnome-keyring libkf5wallet-bin
  gir1.2-gnomekeyring-1.0 python-secretstorage-doc python-setuptools-doc
The following NEW packages will be installed:
  python3-asn1crypto python3-cffi-backend python3-crypto python3-cryptography
[... 154 lines cut ...]

Cleaning up...
Finished! OctoPrint should be ready to go
Once you have verified the install works, you can safely remove the folder /home/pi/oprint.bak
If you want to go back (If it doesn't work) to Python 2 download the file at:
https://raw.githubusercontent.com/cp2004/Octoprint-Upgrade-To-Py3/master/go_back.py
pi@octopi:~ $ ^C
pi@octopi:~ $ OctoPrint successfully installed!
-bash: OctoPrint: command not found
pi@octopi:~ $ Required dependency requests is missing... No plugins can be installed
-bash: Required: command not found
pi@octopi:~ $ OctoPrint has been installed, but no plugins have
-bash: OctoPrint: command not found
pi@octopi:~ $ ^C
pi@octopi:~ $ python3 go_back.py
OctoPrint upgrade to Python 3: go_back.py (v1.1)
This script will move your old installation back (Just in case!)
Only use it if you have used the upgrade script and it failed
Warning: There have been reports of this script failing, if it fails for you please report it to me as soon as possible. Thanks!
Press [enter] to continue, or ctrl-c to stop

Detected OctoPi installation
Successfully reverted to the old install
Before reverting another failed install you should remove the folder /home/pi/oprintFAIL.bak
pi@octopi:~ $ Required dependency requests is missing... No plugins can be installed

cp2004, 2020-10-17:
After installing pip, you need to install requests. Requests should be preinstalled on OctoPi (it is for everyone else, but maybe something went wrong for you. 

Now pip is installed, you can go back and run `python3 -m pip install requests`, then the script, and it should be OK. Still unsure as to why requests is not installed in the first place.

I will add another pre-flight check that requests exists, and directions to install it, so you don't have to proceed only to find out when it tries to download the plugin repository.

wilbo70, 2020-10-17:
Thanks! That did the trick! 
Indeed odd that it wasn't installed?  
I followed the link for bed-visualizer and installed  libatlas3-base, after which this also worked again! 

Have a good weekend!

closed 2020-10-17 by wilbo70