cp2004/Octoprint-Upgrade-To-Py3 #30: [Bug] Describe your bug

leoheck, 2020-10-16:
It is not a bug, it is a question.
If I don't use virtual env, which path should I add there? Nothing seems to work.

cp2004, 2020-10-16:
Not supported by the script unfortunately. I recommend you backup your instance (using built in feature to OctoPrint), then install OctoPrint on your system Python 3, then restore. Although virtualenvs make it quite a lot easier to manage.

closed 2020-10-16 by cp2004

leoheck, 2020-10-17:
Perfect, thank you for the instructions. 
What do you think about adding a note like this on README? It will help users like me.

cp2004, 2020-10-17:
There is now a [Supported Platforms](https://github.com/cp2004/Octoprint-Upgrade-To-Py3#supported-platforms) section in the Readme, that should hopefully make this clear.