cp2004/Octoprint-Upgrade-To-Py3 #32: Upgrade failed to do anything at all...

Beeblebrox666, 2020-10-17:
Script did not work... Why can't this be updated from the octoprint interface like the plugins??? This is difficult for a non-linux person. I do not understand this stuff. trying to look up how to install this python update created 10 more questions and no answers that make any sense.

[image]

jneilliii, 2020-10-18:
I'll make it easier for you...create a manual backup in OctoPrint's settings, and download and install the nightly build of octopi and flash that on to the pi instead. When first booting up supply your backup zip file you downloaded from before.

cp2004, 2020-10-18:
It's architecturally impossible for OctoPrint to perform this update itself, since it is effectively pulling the floor out from under your own feet.

The version of python 3 available on OctoPi 0.16 is not compatible with OctoPrint - so it would not install. So you do need to take a backup, get a new image (as linked by jneilliii) and then restore it.

These instructions are linked from the script, on the homepage: https://github.com/cp2004/Octoprint-Upgrade-To-Py3#what-do-i-do-if-my-system-is-not-supported

closed 2020-10-18 by cp2004