cp2004/Octoprint-Upgrade-To-Py3 #33: [Bug] AttributeError: 'BackupPlugin' object has no attribute '_logger'

bsimmo, 2020-11-12:
**Describe the bug**

**Script output**
Please paste the ***complete*** output of the script in the box below:
```
pi@octopi:~ $ curl -L https://get.octoprint.org/py3/upgrade.py --output upgrade.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   154  100   154    0     0    729      0 --:--:-- --:--:-- --:--:--   729
100 25622  100 25622    0     0  46082      0 --:--:-- --:--:-- --:--:-- 46082
pi@octopi:~ $   python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.1.11)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint (1.4.2) and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
OctoPi version: 0.17.0
Getting OctoPrint version...
OctoPrint version: 1.5.0rc1
Reading installed plugins...
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/__main__.py", line 8, in <module>
    octoprint.main()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/__init__.py", line 939, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup/__init__.py", line 504, in backup_command
    logger=self._logger,
AttributeError: 'BackupPlugin' object has no attribute '_logger'
ERROR: Failed to create OctoPrint backup
Fatal error, exiting
```

**Script version**
OctoPrint Upgrade to Py 3 (v2.1.11)

**OctoPrint Version**
OctoPrint 1.5.0rc1

**Relevant plugins that may have caused errors**
BackupPlugin ?

**Additional information**
I'm just giving feedback on trying to switch to Python3 (Python3.7.3 in this case)

cp2004, 2020-11-12:
@foosel I'll try and reproduce later, but until then is this a bug in 1.5.0rc1?

foosel, 2020-11-12:
Probably - I had to change some stuff around with the backup plugin, so it's likely something that crept in then.

cp2004, 2020-11-12:
Will add some quick detection to the script, and warn of failure using 1.5.0rc1 for now. @bsimmo thanks for reporting this, this is why we have RCs!

closed 2020-11-12 by cp2004

bsimmo, 2020-11-12:
No problem, it's why I reported it.

I'll leave my move to Python3 so I can test against rc2

commit 9ccaab48 (2020-11-12): Add detection of bug in OctoPrint 1.5.0rc1 that prevents using CLI backup

Also version is 2.1.12
Closes #33