cp2004/Octoprint-Upgrade-To-Py3 #36: Script doesn't weork if octoprint user (pi) doesn't have Superuser priveleges [Bug]

heinzz-da-ketchup, 2021-01-02:
**Describe the bug**
On my setup, user `pi` has only specific sudo privileges (`systemctl start/stop/restart octoprint.service`). Script asks to install python3-dev, but doesn't check if it is already installed. If i try to run the script, i can't proceed past sudo prompt for `sudo apt-get update` in function `install_python3_dev`

**Script version**
v2.1.12

**OctoPrint Version**
OctoPrint 1.5.2

**Additional information**
I have added function `check_python3_dev` that uses `dpkg-query -l` to list installed packages and checks if output contains `python3-dev`. It replaces `install_python3_dev` in program flow, and calls that function only if the package is not present. It fixed my (and maybe others problem) and i believe it doesn't break anything.  I will try to make a PR with my fix, i'm not too familiar with Github, so i hope i'll manage to do it, tho. = D

cp2004, 2021-01-02:
Interesting - when you try and run `sudo` commands, there is not even a password prompt? I didn't really think that anyone would have this option. Using `dpkg-query -l` sounds like a better option than running the install (which doesn't do anything if installed) anyway, so I'd be happy to accept that as a contribution 🙂

heinzz-da-ketchup, 2021-01-02:
Hi, to clarify, it pops password prompt, but my `pi` user doesn't even have a password set. = ) I believe that it's a good practice to limit user running any web-facing service to the bare minimum. 
please check the attached PR, i hope i have done everything correctly. = ) My fixed instance upgraded to pyxthon3 correctly.

commit a14e919f (2021-01-03): Merge pull request #37 from heinzz-da-ketchup/devel

:bug: Fix #36 by checking if python3-dev is already installed
commit fd774129 (2021-01-03): :bug: Fix #36 by checking if python3-dev is already installed
closed 2021-01-03 by cp2004