cp2004/Octoprint-Upgrade-To-Py3 #39: Upgrade script for Octoprint with Python3 fail

bendisco51, 2021-06-20:
OS : Debian 8.3.0-6 "Jessie"
Kernel : Linux srvlinux 4.19.0-17-amd64 #1 SMP Debian 4.19.194-1 (2021-06-10) x86_64 GNU/Linux [10.10]
PC : Intel NUC 10i3FNK

SHELL :

root@srvlinux:/home/octoprint# su - octoprint -c "python3 upgrade.py"
OctoPrint Upgrade to Py 3 (v2.1.13)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
Please provide the path to your virtual environment and the config directory of OctoPrint
On OctoPi, this would be `/home/pi/oprint` (venv), `/home/pi/.octoprint` (config)
and service commands commands `sudo service octoprint stop/start`
Path: /opt/OctoPrint/venv
Path valid
Config directory: /opt/OctoPrint/.octoprint
Config directory valid

To do the install, we need the service stop and start commands. (Leave blank if you don't have a service set up)
Stop command: sudo service octoprint stop
Start command: sudo service octoprint start
Getting OctoPrint version...
OctoPrint version: 1.6.1
Reading installed plugins...
2021-06-20 15:25:42,552 - octolapse.__init__ - INFO - Release mode detected.
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/opt/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 31, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/opt/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 31, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
Failed to read created backup & plugins list
If you ran this script using root (`sudo`), please make sure you don't, and run using `python3 upgrade.py`, as per the guide.
The other reason this may happen is if you are not running as the same user OctoPrint is installed as/runs under.
Error: Could not read backup

cp2004, 2021-06-22:
Hey @bendisco51! Apologies for the delay in replying, I initially didn't have any ideas until this evening....

You run with a custom base directory location... I was typing out some tests for you but then remembered what is the issue. Do you usually run the server with the `--basedir` flag (`=/opt/OctoPrint/.octoprint`)?

The script is looking for the backup in `/opt/OctoPrint/.octoprint`, as you told it, but it doesn't pass `--basedir` when it is *creating* the backup, so it is probably in `/home/octoprint/.octoprint/...` instead.

Try again with my test fix, from this URL instead of `get.octoprint.org`:
```
https://raw.githubusercontent.com/cp2004/Octoprint-Upgrade-To-Py3/issue39/upgrade.py
```
The script version should be `2.1.14-dev1`, when you download it. ⚠️ This is a development version, take a backup and know how to restore that backup first ⚠️

bendisco51, 2021-06-23:
Hello @cp2004, no problem for the delay, upgrade to python3 is not urgent.

Here is some additional info :

**nano /etc/default/octoprint**
`# Configuration for /etc/init.d/octoprint

# The init.d script will only run if this variable non-empty.
OCTOPRINT_USER=octoprint

# base directory to use
#BASEDIR=/home/pi/.octoprint
BASEDIR=/opt/OctoPrint/.octoprint

# configuration file to use
#CONFIGFILE=/home/pi/.octoprint/config.yaml

# On what port to run daemon, default is 5000
PORT=5000

# Path to the OctoPrint executable, you need to set this to match your installation!
#DAEMON=/home/pi/OctoPrint/venv/bin/octoprint
DAEMON=/opt/OctoPrint/venv/bin/octoprint

# What arguments to pass to octoprint, usually no need to touch this
DAEMON_ARGS="--port=$PORT"

# Umask of files octoprint generates, Change this to 000 if running octoprint as its own, separate user
UMASK=022

# Process priority, 0 here will result in a priority 20 process.
# -2 ensures Octoprint has a slight priority over user processes.
NICELEVEL=-2

# Should we run at startup?
START=yes
`

**ls /home/octoprint**
`ls: impossible d'accéder à '/home/octoprin': Aucun fichier ou dossier de ce type`
(don't exist)

**ls /opt/OctoPrint/.octoprint**
`config.backup  data       logs     printerProfiles  slicingProfiles  translations  users.yaml  watched
config.yaml    generated  plugins  scripts          timelapse        uploads       virtualSd
`

I can try your patched script if you think is good with my install ?
Thanks,
Benoit

cp2004, 2021-06-23:
The patched script should work for you.

bendisco51, 2021-06-23:
I have tried your patched script. I have some error but the upgrade is fine.  Thanks !

`
OctoPrint Upgrade to Py 3 (v2.1.14-dev1)

Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
Please provide the path to your virtual environment and the config directory of OctoPrint
On OctoPi, this would be `/home/pi/oprint` (venv), `/home/pi/.octoprint` (config)
and service commands commands `sudo service octoprint stop/start`
Path: /opt/OctoPrint/venv
Path valid
Config directory: /opt/OctoPrint/.octoprint
Config directory valid

To do the install, we need the service stop and start commands. (Leave blank if you don't have a service set up)
Stop command: sudo systemctl stop octoprint.service
Start command: sudo systemctl start octoprint.service
Getting OctoPrint version...
OctoPrint version: 1.6.1
Reading installed plugins...
2021-06-23 15:20:45,145 - octolapse.__init__ - INFO - Release mode detected.
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/opt/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 31, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/opt/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 31, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'

Plugins installed
- Backup Scheduler
- PrintTimeGenius Plugin
- Octolapse
If there is an error above related to OctoLapse, please ignore, it makes no difference to operation :)
If you think there is something missing from here, please check the list of plugins in Octoprint
Press [enter] to continue or ctrl-c to quit

Checking package list for python3-dev
Successfully checked python3-dev
Creating new Python 3 environment...
/usr/lib/python2.7/dist-packages/virtualenv.py:1090: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Successfully created Python 3 environment at /opt/OctoPrint/venv

Installing OctoPrint... (This may take a while - Do not cancel!)
Collecting required packages
OctoPrint successfully installed!

Downloading OctoPrint's plugin repo

Installing Octolapse
Collecting required packages
Installing collected packages
Plugin Octolapse successfully installed
Installing OctoPrint-PrintTimeGenius
Collecting required packages
Installing collected packages
Plugin OctoPrint-PrintTimeGenius successfully installed
Installing Backup Scheduler
Collecting required packages
Installing collected packages
Plugin Backup Scheduler successfully installed

Cleaning up...
Finished! OctoPrint should be ready to go
Once you have verified the install works, you can safely remove the folder /opt/OctoPrint/venv.bak
If you want to go back (If it doesn't work) to Python 2 download the file at:
https://raw.githubusercontent.com/cp2004/Octoprint-Upgrade-To-Py3/master/go_back.py
`

cp2004, 2021-06-23:
As it says:
> If there is an error above related to OctoLapse, please ignore, it makes no difference to operation :)

It's a problem with the OctoLapse plugin, and is handled & makes no difference to the backup or the upgrade script. Great to know we got this sorted, I will push the fix out to the 'production' version :)

closed 2021-06-23 by cp2004