cp2004/Octoprint-Upgrade-To-Py3 #41: Script it doesn't work here

robustini, 2021-06-26:
**Describe the bug**
Unfortunately I'm not using OctoPi but it's still a Buster.
The error is in the output. 

**Script output**
Hello!
This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Detecting system info...
Please provide the path to your virtual environment and the config directory of OctoPrint
On OctoPi, this would be `/home/pi/oprint` (venv), `/home/pi/.octoprint` (config)
and service commands commands `sudo service octoprint stop/start`
Path: /home/pi/OctoPrint
Invalid venv path, please try again
Path: /home/pi/OctoPrint/venv
Path valid
Config directory: /home/pi/.octoprint
Config directory valid

To do the install, we need the service stop and start commands. (Leave blank if you don't have a service set up)
Stop command: sudo service octoprint stop
Start command: sudo service octoprint start
Getting OctoPrint version...
OctoPrint version: 1.6.1
Reading installed plugins...
2021-06-26 19:27:55,231 - octolapse.__init__ - INFO - Release mode detected.
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1669, in shutdown
    h.close()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 31, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1669, in shutdown
    h.close()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 31, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'

Plugins installed
- DisplayLayerProgress Plugin
- DryRun Plugin
- Firmware Updater
- PrettyGCode
- Fullscreen Plugin
- PSU Control - RPi.GPIO
- Custom Control Editor
- FileManager
- Autoscroll
- OctoKlipper
- Fan Speed Control
- InlineConfirm
- System Command Editor
- Detailed Progress
- Dashboard
- PSU Control
- Arc-Welder
- Octolapse
If there is an error above related to OctoLapse, please ignore, it makes no difference to operation :)
If you think there is something missing from here, please check the list of plugins in Octoprint
Press [enter] to continue or ctrl-c to quit

Checking package list for python3-dev
Successfully checked python3-dev
Creating new Python 3 environment...
ERROR: Failed to create Python 3 venv
Could not create new venv

Cleaning up...
Fatal Error: Exiting
pi@Raspberry1:~ $

**Script version**
The latest of today.

**OctoPrint Version**
OctoPrint 1.6.1

**Relevant plugins that may have caused errors**
I don't think plugins are causing any problems. 

**Additional information**
None.

cp2004, 2021-06-26:
What happens if you run the command `python -m virtualenv --python=/usr/bin/python3 /home/pi/OctoPrint/venv_test`?

cp2004, 2021-06-26:
(Side note) I am aware, that that command above is not very OS-agnostic, so if that doesn't work, try this one:
```
python3 -m venv /home/pi/OctoPrint/venv_test
```

robustini, 2021-06-26:
> python3 -m venv /home/pi/OctoPrint/venv_test

Thanks for the reply!
The new venv is created without problems. 

pi@Raspberry1:~ $ cd OctoPrint/
pi@Raspberry1:~/OctoPrint $ ls
venv  venv_test
pi@Raspberry1:~/OctoPrint $

robustini, 2021-06-26:
> 
> 
> (Side note) I am aware, that that command above is not very OS-agnostic, so if that doesn't work, try this one:
> 
> ```
> python3 -m venv /home/pi/OctoPrint/venv_test
> ```

Thanks for the reply!
The new venv is created without problems.

[image]

robustini, 2021-06-26:
When the script fails it renames the current venv to venv.bak if this is useful for you, but the new one is not created.

cp2004, 2021-06-26:
Which command have you used to create that? I see you quoted the reply with the alternative one, just wanted to confirm that the first one failed - if the second works, I think it makes more sense to make the script use that one instead.

robustini, 2021-06-26:
> Which command have you used to create that? I see you quoted the reply with the alternative one, just wanted to confirm that the first one failed - if the second works, I think it makes more sense to make the script use that one instead.

I use the second.

cp2004, 2021-06-26:
Please try the first, and report back whatever error you find, as I said *if that doesn't work, try this one*

robustini, 2021-06-27:
> 
> 
> Please try the first, and report back whatever error you find, as I said _if that doesn't work, try this one_

The result:

/usr/local/bin/python3.9: No module named virtualenv

Then I install virtualenv :

 pip3 install virtualenv
 
 And when I try to create again venv:
 
 FileNotFoundError: [Errno 2] No such file or directory: '/usr/bin/python3'

robustini, 2021-06-27:
By modifying your script with the working command for the creation of the venv then the result is this:

Checking package list for python3-dev
Successfully checked python3-dev
Creating new Python 3 environment...
Successfully created Python 3 environment at /home/pi/OctoPrint/venv

Installing OctoPrint... (This may take a while - Do not cancel!)
Collecting required packages
Installing collected packages
Collecting required packages

Installing collected packages
WARNING: You are using pip version 21.1.2; however, version 21.1.3 is available.
You should consider upgrading via the '/home/pi/OctoPrint/venv/bin/python -m pip install --upgrade pip' command.
OctoPrint successfully installed!
Required dependency requests is missing... No plugins can be installed
OctoPrint has been installed, but no plugins have

Cleaning up...
Finished! OctoPrint should be ready to go
Once you have verified the install works, you can safely remove the folder /home/pi/OctoPrint/venv.bak
If you want to go back (If it doesn't work) to Python 2 download the file at:
https://raw.githubusercontent.com/cp2004/Octoprint-Upgrade-To-Py3/master/go_back.py

robustini, 2021-06-27:
You should consider upgrading via the '/home/pi/OctoPrint/venv/bin/python -m pip install --upgrade pip' command.

Somehow your script invokes Python version 2.7 when trying to install plugins, or there is something misconfigured in my file system.

robustini, 2021-06-27:
You should consider upgrading via the '/home/pi/OctoPrint/venv/bin/python -m pip install --upgrade pip' command.

There is something misconfigured in my file system?
I don't understand why if in this way it installs Octoprint correctly then it doesn't install the plugins, maybe because once installed it doesn't find the updated version of pip?

robustini, 2021-06-27:
This is the content of the venv after go back:

pi@Raspberry1:~/OctoPrint/venvFAIL.bak $ ls -l
total 12
drwxr-xr-x 3 pi pi 4096 Jun 27 09:25 bin
drwxr-xr-x 3 pi pi 4096 Jun 27 09:23 lib
-rw-r--r-- 1 pi pi  225 Jun 27 09:23 pyvenv.cfg
pi@Raspberry1:~/OctoPrint/venvFAIL.bak $

cp2004, 2021-06-27:
> WARNING: You are using pip version 21.1.2; however, version 21.1.3 is available.
You should consider upgrading via the '/home/pi/OctoPrint/venv/bin/python -m pip install --upgrade pip' command.

This is a warning directly from pip, not the script, I have no control over this.

> ```
> Required dependency requests is missing... No plugins can be installed
> ```
This means the `python3-requests` package is not installed on your system - it has nothing to do with the pip warning above. It is generally available in most OS installs, and it is needed to download the plugin repository list to find all the install locations as they are not on PyPi, and have to be installed using a URL instead. I haven't had an issue with people not having this package before, but there is error handling (as you found) to help try and explain what happened. I should probably make the warning show at the beginning of the script as well, so you know *before* you perform the upgrade.

I will test and release an update in the next few days that helps with some of these issues, now that we have dug into the causes. It's very difficult to work for *everyone*, so outside of OctoPi installs this is not well tested.

robustini, 2021-06-27:
Thanks for the explanation!
"pip3 install requests" and now everything works!

closed 2021-06-27 by robustini

skom27, 2021-12-19:
I'm trying on Orange Pi (Ubuntu 16.04.7 LTS and OctoPrint 1.7.2) to upgrade to python ver. 3 with python 2.7. When executing upgrade.py I get error: E: Command line option --allow-releaseinfo-change is not understood in combination with the other options
`octoprint@OrangePi:$ python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.2.2)

This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Checking system info...
Please provide the path to your virtual environment and the config directory of OctoPrint
On OctoPi, this would be Yhome/pi/oprint' (for the venv), '/home/pi/.octoprint' (for config)
and service commands commands 'sudo service octoprint stop/start'
Virtual environment path: /home/octoprint/OctoPrint/venv
Found version: Python 2.7.12
Path valid
Config directory path: /home/octoprint/.octoprint
Config directory valid

To do the install, we need the service stop and start commands. (Leave blank if
you don't have a service set up)
Stop command: sudo service octoprint stop
Start command: sudo service octoprint start
Checking OctoPrint version...
OctoPrint version: 1.7.2
Reading installed plugins...

Plugins installed
- DisplayLayerProgress Plugin
- Slicer
- Sidebar Webcam
- STL Viewer
- Bed Visualizer
- GcodeEditor
- Detailed Progress
- OctoPrint Anywhere
- Dashboard
- Autoscroll
If you think there is something missing from here, please check the list of plugins in OctoPrint
Press [enter] to continue or ctrl-c to quit
Updating package list...
E: Command line option --allow-releaseinfo-change is not understood in combination with the other options

Please try manually

Cleaning up...

octoprint@OrangePi:$ I

--------------------------------------------------------------------------

orangepi@OrangePi:~$ python
Python 2.7.12 (default, Mar  1 2021, 11:38:31)
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>>

--------------------------------------------------------------------------

orangepi@OrangePi:~$ python3
Python 3.8.9 (default, Apr  3 2021, 01:02:10)
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>

`