cp2004/Octoprint-Upgrade-To-Py3 #43: Error: Failed to create Python 3 venv

kidkill, 2021-07-05:
Could not create new venv                                                                                                                                                                                                                           Cleaning up...                                                                             
Fatal Error: Exiting 

**Script output**
Please paste the ***complete*** output of the script in the box below:
```
OctoPrint Upgrade to Py 3 (v2.1.15)                                                                                                                                                                                                                 Hello!                                                                                                                    This script will move your existing OctoPrint configuration from Python 2 to Python 3                                     This script requires an internet connection and it will disrupt any ongoing print jobs.                                                                                                                                                             It will install the latest version of OctoPrint and all plugins.                                                          No configuration or other files will be overwritten                                                                                                                                                                                                 Press [enter] to continue or ctrl-c to quit                                                                                                                                                                                                         Detecting system info...                                                                                                  OctoPi version: 0.17.0                                                                                                    Getting OctoPrint version...                                                                                              OctoPrint version: 1.6.1                                                                                                  Reading installed plugins...                                                                                                                                                                                                                        Plugins installed                                                                                                         - Filament Sensor Reloaded                                                                                                - OctoPrint-IFTTT                                                                                                         - Filament Scale                                                                                                          - Enclosure Plugin                                                                                                        - GPIO Shutdown                                                                                                           If you think there is something missing from here, please check the list of plugins in Octoprint                          Press [enter] to continue or ctrl-c to quit                                                                                                                                                                                                         Checking package list for python3-dev                                                                                     python3-dev not installed, proceeding to install                                                                          Root access is required to install python3-dev, please fill in the password prompt if shown                               Updating package list...                                                                                                  [sudo] password for pi:                                                                                                   Installing python3-dev...                                                                                                 Successfully installed python3-dev                                                                                        Creating new Python 3 environment...                                                                                      ERROR: Failed to create Python 3 venv                                                                                     Could not create new venv                                                                                                                                                                                                                           Cleaning up...                                                                                                            Fatal Error: Exiting                                                                                                      pi@octopi:~ $
```

**Script version**
You can find this printed in the first line of the script

**OctoPrint Version**
OctoPrint 1.().()

**Relevant plugins that may have caused errors**
If you're having trouble with a particular part, please include your installed plugins

**Additional information**
Screenshots, links to docs, etc.

cp2004, 2021-07-05:
Could you try running `python3 -m venv test-venv` and report the results?

ElDavoo, 2021-07-05:
Creating new Python 3 environment...
The virtual environment was not created successfully because ensurepip is not

available.  On Debian/Ubuntu systems, you need to install the python3-venv

package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv

package, recreate your virtual environment.

Failing command: ['/home/pi/oprint/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']

You need to install python3-venv and it will work.

I'm using latest octoprint and octopi 0.17.0.

Documentation should be updated.

cp2004, 2021-07-05:
This was a recent change - #42 - I will revert it, the previous one worked on more systems apparently.

A smarter handling of the failures that made me want to switch it is probably smart.

cp2004, 2021-07-05:
Thanks for reporting the error @ElDavoo - I have reverted the changes. It may take a short while for caching to expire, the new script version is 2.1.16. Just in case you felt like using it again.

closed 2021-07-05 by cp2004

ElDavoo, 2021-07-05:
It's still installing plugins right now because of #44 :)