cp2004/Octoprint-Upgrade-To-Py3 #44: wheel is not automatically installed

ElDavoo, 2021-07-05:
The install script does not automatically install the wheel package for debian/pip, resulting in "fake" errors that can scary people out and slow plugin install times. Note that octoprint and plugins do install correctly with the traditional setup.py install.

**Script output**
For every package:
```
Failed building wheel for Octolapse
  error: invalid command 'bdist_wheel'
```

**Script version**
You can find this printed in the first line of the script  
It's not actually the first line, but it's 2.1.16

**OctoPrint Version**
OctoPrint 1.6 kind of

cp2004, 2021-07-05:
> It's not actually the first line, but it's 2.1.16

Well first line of the *script output*... Anyway

This makes sense to include, thank you for pointing it out.

cp2004, 2021-08-07:
Wheel is installed with the latest version, which again drops the dependency on Python 3 by installing `python3-venv` properly this time. Not that you really care since you have already upgraded to Python 3, but I have to close this issue with some comment... Script version will be 2.2.0, once I have merged it to master.

closed 2021-08-07 by cp2004

Kiina, 2021-08-31:
I have `SCRIPT_VERSION = '2.2.0'` but it still gives me tons of errors related to wheel

```sh
Installing Navbar Temp
Collecting required packages
  Failed building wheel for OctoPrint-NavbarTemp
Error installing package
  Running setup.py bdist_wheel for OctoPrint-NavbarTemp: finished with status 'error'

Error installing package
  error: invalid command 'bdist_wheel'

Installing collected packages
Plugin Navbar Temp successfully installed
```

I mean it worked fine in the end but seeing a red terminal for 10 minutes doesn't really make you feel confident the script is running correctly.

cp2004, 2021-08-31:
@Kiina the script *should* (and if it did not report an error to you under 'Installing build dependencies') install wheel. In most of the times I tested this, it worked great, I had one time that it didn't. However, you aren't the first person to report this. I don't know why it seems to work sometimes... Maybe there's something else that needs to be installed as well.

ErroneousBosch, 2021-09-24:
I am seeing this exact same issue. I also did not have pip installed (Octoprint 0.17.0), but rolling back, installing pip, and re-running made no difference.

cp2004, 2021-09-24:
That's annoying - I am not sure what else I can do in the script to install it. This issue definitely doesn't affect everyone, and when I compared the results of installing `wheel` vs. not installing, the errors *should* be fixed. Maybe it doesn't get used when trying to install more things straight away. I haven't *really* got time to investigate thoroughly at the moment, but will open another issue to track to hopefully mean we can get this fixed at some point.

ErroneousBosch, 2021-09-25:
Eh, I ended up just flashing a new updated  OctoPi card and transferring my stuff over. Took less time than I was spending futzing with the errors. No worries.

cp2004, 2021-09-25:
Was the script not successful? Despite the errors, it does still install everything successfully and it should work. It just doesn't install using wheel, only the standard install.

ErroneousBosch, 2021-09-27:
It said it was, but I was having connection issues, hence why I just went ahead and did a fresh flash