cp2004/Octoprint-Upgrade-To-Py3 #47: [Bug] Pip errors as wheel is not working properly *sometimes*.

cp2004, 2021-09-24:
See comments in #44

Wheel *should* be automatically installed, and no errors are reported from this process. But sometimes it shows up loads of error messages.

**The upgrade does actually succeed**. Pip is smart enough to work around the issues and get you running any way possible. Please don't panic, but if you know what you are looking at please think of any possible solutions and let me know. I don't know what to do here and have very little time for a proper investigation dive into it at the moment.

Lumanet2012, 2021-10-15:
To whom it may concern,

I ran also the upgrade script and did ran into issues but so far it finished successfully under 1.7.0 with 0.17 and installed at the end python 3.7.3 and today is October 15th 2021. 

attached is a full log of what I did and so far the outcome is that it look fully functional.

[Python upgrade on 0.17 and 1.7.0 from python 2 to 3 via script.txt](https://github.com/cp2004/Octoprint-Upgrade-To-Py3/files/7353853/Python.upgrade.on.0.17.and.1.7.0.from.python.2.to.3.via.script.txt)

humanoid2050, 2021-10-16:
I think I figured out the issue here and with #44. Line 536 of the update script does not have square braces to indicate that the args to run_sys_command are all part of a single command. As such, pip gets invoked by the system without args, and that is a perfectly valid invocation. Checking the return code yields 0 and the script moves on. Unfotunately, the "install" and "wheel" args get routed to the 'custom_parser' and 'sudo' arguments of run_sys_command.

closed 2021-10-16 by cp2004