cp2004/Octoprint-Upgrade-To-Py3 #58: [Bug] After seemingly successful update to python 3, "Software Update" is disabled

ross, 2021-11-06:
**Describe the bug**

After following https://octoprint.org/blog/2020/09/10/upgrade-to-py3/ "Software Update" is blocked b/c of unexpected versions.

[image]

It's showing lots of updates available, but the button is disabled:

[image]

Everything seems to work fine. I'm able to print and haven't run across anything else broken or unhappy. I'll probably look at a fresh install and import my config, but figured I'd file this and give a bit of time to see if there's any information you all would like me to collect or anything you'd like me to try out/dig into.

**Script output**

I didn't see anything noteworthy during the process and I've long since closed the terminal I ran it in. If the log is on disk somewhere lmk and I can grab it. It wasn't until today (about a week later) that I I got the updates available popup and went to update, but couldn't.

**Script version**

Same

**OctoPrint Version**

OctoPrint 1.7.0

cp2004, 2021-11-06:
About 5 days ago I made it update pip as well, this should no longer be an issue.

Have you tried clicking on the link that gives you instructions of how to update pip to a newer version? Full reinstalling is unnecessary and can be avoided by following some instructions.

ross, 2021-11-06:
Thanks @cp2004 for the quick reply. 

> Have you tried clicking on the link that gives you instructions of how to update pip to a newer version? Full reinstalling is unnecessary and can be avoided by following some instructions.

Didn't look for anything pip specific since it was also complaining about having python 3 when it expected python 2. 

Just updated pip in the oprint venv:

```
pi@octopi:~ $ source ./oprint/bin/activate
(oprint) pi@octopi:~ $ pip install --upgrade pip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Cache entry deserialization failed, entry ignored
Cache entry deserialization failed, entry ignored
Collecting pip
  Downloading https://files.pythonhosted.org/packages/a4/6d/6463d49a933f547439d6b5b98b46af8742cc03ae83543e4d7688c2420f8b/pip-21.3.1-py3-none-any.whl (1.7MB)
    100% |████████████████████████████████| 1.7MB 169kB/s
Installing collected packages: pip
  Found existing installation: pip 18.1
    Uninstalling pip-18.1:
      Successfully uninstalled pip-18.1
Successfully installed pip-21.3.1
(oprint) pi@octopi:~ $ pip --version
pip 21.3.1 from /home/pi/oprint/lib/python3.7/site-packages/pip (python 3.7)
```

Still get the message in the "Software Update" tab, but it hasn't changed the pip version there either so I'm guessing it's stale/cached. Check for updates doesn't seem to redo it so I'll try a OctoPrint restart after my current print finishes.

cp2004, 2021-11-06:
> it was also complaining about having python 3 when it expected python 2.

They're minimum versions - Python 3 is above 2.

It requires at least an OctoPrint restart, and sometimes a force-check to override the cache - under advanced options.

ross, 2021-11-06:
Cool. Restart of OctoPrint and a force check for update as suggested cleared up the problem.

closed 2021-11-06 by ross