cp2004/Octoprint-Upgrade-To-Py3 #61: [Bug]

Wimll, 2021-11-13:
**Describe the bug**

Updating octoprint to pyhton 3 failed, now the server is not starting and I am not able to ssh to it

**Script output**
Please paste the ***complete*** output of the script in the box below:
```
pi@octopi:~ $ python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.2.2)

This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Checking system info...
Detected OctoPi version 0.17.0
Found version: Python 2.7.16
Checking OctoPrint version...
OctoPrint version: 1.7.2
Reading installed plugins...

Plugins installed
- Tasmota
- Telegram Notifications
- Draggable Files
- Access Anywhere - The Spaghetti Detective
- DisplayZ
- BetterHeaterTimeout
- MeatPack
- Autoselect Plugin
- M73 Progress Plugin
- Dragon Order
- Tab Order
- Themeify
- Autoscroll
- Fullscreen Plugin
- TouchUI
- OctoEverywhere!
- Bed Visualizer
- Simple Emergency Stop
- Backup Scheduler
- Navbar Temperature Plugin
- Marlin EEPROM Editor
- Preheat Button
If you think there is something missing from here, please check the list of plugins in OctoPrint
Press [enter] to continue or ctrl-c to quit

Updating package list...

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

[sudo] password for pi:

Checking package list for python3-dev
Installing python3-dev...
Successfully installed python3-dev
Checking package list for python3-venv
Installing python3-venv...
Successfully installed python3-venv
Creating new Python 3 environment...
Successfully created Python 3 environment at /home/pi/oprint
Installing build dependencies...
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
Updating pip...
Cache entry deserialization failed, entry ignored
Cache entry deserialization failed, entry ignored

Installing OctoPrint... (This may take a while - Do not cancel!)
Collecting required packages
Installing collected packages
OctoPrint successfully installed!

Downloading OctoPrint's plugin repo

Installing Navbar Temp
Collecting required packages
Installing collected packages
Plugin Navbar Temp successfully installed
Installing Autoselect
Collecting required packages
Installing collected packages
Plugin Autoselect successfully installed
Installing Marlin EEPROM Editor
Collecting required packages
Installing collected packages
Plugin Marlin EEPROM Editor successfully installed
Installing DisplayZ
Collecting required packages
Installing collected packages
Plugin DisplayZ successfully installed
Installing TouchUI
Collecting required packages
Installing collected packages
Plugin TouchUI successfully installed
Installing OctoPrint-Telegram
Collecting required packages
Installing collected packages
Plugin OctoPrint-Telegram successfully installed
Installing Autoscroll
Collecting required packages
Installing collected packages
Plugin Autoscroll successfully installed
Installing Fullscreen Webcam
Collecting required packages
Installing collected packages
Plugin Fullscreen Webcam successfully installed
Installing Simple Emergency Stop
Collecting required packages
Installing collected packages
Plugin Simple Emergency Stop successfully installed
Installing Themeify
Collecting required packages
Installing collected packages
Plugin Themeify successfully installed
Installing Preheat Button
Collecting required packages
Installing collected packages
Plugin Preheat Button successfully installed
Installing OctoPrint-Tasmota
Collecting required packages
Installing collected packages
Plugin OctoPrint-Tasmota successfully installed
Installing M73 Progress
Collecting required packages
Installing collected packages
Plugin M73 Progress successfully installed
Installing OctoPrint-TabOrder
Collecting required packages
Installing collected packages
Plugin OctoPrint-TabOrder successfully installed
Installing Bed Level Visualizer
Collecting required packages
Installing collected packages
Plugin Bed Level Visualizer successfully installed
Warning: You have installed Bed Level visualiser. There is a known issue with it failing silently on Python 3
See more here: https://github.com/jneilliii/OctoPrint-BedLevelVisualizer#known-issues
Installing OctoPrint-DragonOrder
Collecting required packages
Installing collected packages
Plugin OctoPrint-DragonOrder successfully installed
Installing OctoPrint-BetterHeaterTimeout
Collecting required packages
Installing collected packages
Plugin OctoPrint-BetterHeaterTimeout successfully installed
Installing Access Anywhere - The Spaghetti Detective
Collecting required packages
Error installing package
  Building wheel for TheSpaghettiDetective (setup.py): finished with status 'error'

Connection to octopi.local closed by remote host.
Connection to octopi.local closed.
```

**Script version**
You can find this printed in the first line of the script

**OctoPrint Version**
OctoPrint 1.7.2

**Relevant plugins that may have caused errors**
If you're having trouble with a particular part, please include your installed plugins

**Additional information**
Screenshots, links to docs, etc.

Any idea what I can do to restore my octoprint? I am not able to acces it now, even going back to python 2 will be fine for me

cp2004, 2021-11-14:
If you can no longer access the Pi over SSH (presuming you have tried rebooting it?) then you probably have to connect it up to a HDMI monitor to try and figure out why it is no longer booting. Looks like maybe the Pi crashed during the install somehow?

Wimll, 2021-11-14:
> If you can no longer access the Pi over SSH (presuming you have tried rebooting it?) then you probably have to connect it up to a HDMI monitor to try and figure out why it is no longer booting. Looks like maybe the Pi crashed during the install somehow?

Thanks! I plugged it into HDMi, the Pi doesn't boot properly and shows 'Kernel panic: Unable to mount root fs on unknown block'

Is there any way to save the footprint backups before I reflash octoprint?

cp2004, 2021-11-14:
Sounds like something on the SD card might have died, not really sure. If you are lucky, you might be able to find something else that will read it (another linux thing, or on Windows I have 'DiskInternals Linux Reader' which seems to work). But if the SD card has died it might not work.

cp2004, 2021-11-23:
If you still need help upgrading your install to Python 3, please reach out on the [community forums](https://community.octoprint.org) or [discord server](https://discord.octoprint.org) for assistance.

closed 2021-11-23 by cp2004