cp2004/Octoprint-Upgrade-To-Py3 #62: [Bug]

treii28, 2021-11-14:
upgrade won't run properly on non privileged account.

It is not a good idea to run things like octopi on privileged accounts (accounts with root or sudo access). I turn off the privileges of the 'pi' account since it is a typical attack vector used by hackers. Thus if I try to run your script as root, it says it won't run. But if I try to run it as the pi user, it says it wants sudo access. This is bad design. Meanwhile, if I try to run it as my own user, it can't write the pi user directories (again, privilege separation means I 'should not' be able to write pi/server/octoprint directories with a user account without using sudo first)
You need to re-think how you go about that and possibly add a parameter or conditionals to specify the octopi user and/or check that it's installing in a way that can access that user's files as well as do any sudo access you need.

cp2004, 2021-11-14:
This script works for 99% of use cases, I don't see an easy way through to make this work while keeping it as simple as it is for the normal OctoPrint user.

It needs to run as the user OctoPrint is installed as, `pi` in your case so that it can access the installation, the plugins, make a backup and copy the stuff across to the new installation. It also needs to check that a few packages are installed. First it runs `sudo apt update`, then some `dpkg` commands that don't actually require root access to check if it needs to install additional packages. I guess the apt update could be switched and not run if not necessary, and you can just install `python3-dev` and `python3-venv` manually if this problem affects you.

I don't know of a way I can get people to run this script, run some things as one user and the rest as another user without running it as root. Running it as root then means I have to get people to provide the user they have installed OctoPrint as - this is harder than it sounds, as people try and run this script in OctoPrint's terminal for printer communication sometimes - most of them have no idea.

I won't be turning it on it's head and redesigning this script just to fit your use case - it sounds like you understand the process well enough to do it manually, so here's the 5 steps:

1. Take a backup from within OctoPrint's backup/restore page. Can exclude everything possible.
2. Move old virtualenv out of the way with `mv <path to old> <somewhere else` for safekeeping in case it goes wrong
3. Create a new one: `python3 -m venv <path>`
4. Install OctoPrint: `<path>/bin/pip install octoprint`
5. Restore the backup to automatically download the plugins.

cp2004, 2021-11-28:
If you still need help upgrading your install to Python 3, please reach out on the [community forums](https://community.octoprint.org) or [discord server](https://discord.octoprint.org) for assistance.

closed 2021-11-28 by cp2004