cp2004/Octoprint-Upgrade-To-Py3 #63: Command line option --allow-releaseinfo-change is not understood in combination with the other options

ZiniuLu, 2022-01-05:
Hardware: raspberry pi 3B
OS: Raspbian GNU/Linux 9 (stretch)
Environment:
- OctoPrint: 1.7.2
- Python: 2.7.13 (you have 2.7.13)
- pip: 19.0.1 (you have 18.0)
- setuptools: 40.7.1 (you have: 40.2.0)

I have already installed Python3.7.4 manually und updated OctoPrint to 1.7.2. The screenshot below shows what I've done to for `upgrade.py`. And I got this: 

`E: Command line option --allow-releaseinfo-change is not understood in combination with the other options`. 

I think the reason for this error is due to the fact that my system is the `stretch` version. How do I fix this problem without reinstalling my system? 

[image]

cp2004, 2022-01-05:
Probably easiest to edit the script to remove the extra argument - maybe older versions of apt & co don't have that option. I didn't design the script to work on Debian OSes where Python 3.6+ was not available in the repositories.

You can find it on line 193 to remove it.

ZiniuLu, 2022-01-07:
@cp2004 Thanks for your help. Finally I successfully updated OctoPrint's environment to Python 3.7.4. So you can close this issue now. I will document my procedure below in case someone else has encountered similar problems.

Error: `E: Command line option --allow-releaseinfo-change is not understood in combination with the other options`
Operation: removed the argument `--allow-releaseinfo-change` and reran the script 'upgrade.py'

Error: `Failed to  create Python 3 venv` because of `ModuleNotFoundError: no module named apt_pkg`
Error: `Command '('lsb_release', '-a')' returned non-zero exit status 2`
Operations: 
- `cd /usr/lib/python3/dist-packages/`
- `sudo ln -s apt_pkg.cpython-37m-x86_64-linux-gnu.so apt_pkg.so`
- `sudo apt-get upgrade`
- `sudo apt-get --fix-broken install`
- `sudo apt-get remove python3.5*`
- `sudo apt-get autoremove`
- `sudo apt-get --purge remove && sudo apt-get autoclean`
- `cd /var/lib/dpkg/info/`
- `rm the-package-causing error`
- `sudo mv info info.bak`
- `sudo mkdir info`
- `sudo apt-get update`
- `sudo apt-get upgrade && sudo apt-get dist-upgrade`
- `sudo apt-get install python3-dev`
- `sudo apt-get install python3-venv`
Then excute the `upgrade.py` again
- `cd /home/pi/.../OctoPrint/`
- `sudo rm -rf venv`
- `sudo mv venv.bak venv`
- `python3 upgrade.py`

cp2004, 2022-01-07:
Thanks for getting back with the full steps that worked for you to get it running. I'll keep it in mind if I see other issues. Although now it makes it easier to see why I didn't recommend the manual Python installation approach, especially to people who didn't know what they were doing as they wouldn't have been able to figure that out. 👍

closed 2022-01-07 by cp2004