cp2004/Octoprint-Upgrade-To-Py3 #64: Unsigned Repository stops script[Bug]

travisedson, 2022-01-15:
**Describe the bug**
The script trys to draw from an unsigned repository which causes an error.

**Script output**
pi@octoPi:~ $ python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.2.2)

This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Checking system info...
Please provide the path to your virtual environment and the config directory of OctoPrint
On OctoPi, this would be `/home/pi/oprint` (for the venv), `/home/pi/.octoprint` (for config)
and service commands commands `sudo service octoprint stop/start`
Virtual environment path: /home/pi/OctoPrint
Found version: Python 2.7.16
Path valid
Config directory path: /home/pi/.octoprint
Config directory valid

To do the install, we need the service stop and start commands. (Leave blank if you don't have a service set up)
Stop command: sudo service octoprint stop
Start command: sudo service octoprint start
Checking OctoPrint version...
OctoPrint version: 1.4.2
Reading installed plugins...
No plugins found
If you think this is an error, please ask for help. Note this doesn't include bundled plugins.
Press [enter] to continue or ctrl-c to quit

Updating package list...
W: GPG error: https://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease: The following signatures were invalid: 88E8F32F724468BA39585D4099DA5D2AFCE635A4
E: The repository 'http://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease' is not signed.
ERROR: failed to update package list
Please try manually

Cleaning up...
Fatal error: Exiting

**Script version**
SCRIPT_VERSION = '2.2.2'

**OctoPrint Version**
Version 1.7.2

**Relevant plugins that may have caused errors**
I don't think plugins have anything to do with this.

**Additional information**
There's really nothing more to add.

cp2004, 2022-01-15:
This is nothing wrong with the script, something is broken in your apt configuration. It is just trying to run `sudo apt update` to update the package list, if that is failing here it will fail outside the script as well.

travisedson, 2022-01-15:
ok. but i can put a flag in with sudo apt update to let it continue anyways. how can i add that flag to the script? or what can i do to fix this issue. i've been googling around for literally hours getting nowhere.

cp2004, 2022-01-15:
You will have to edit the script to add the flag manually then. Line 193

travisedson, 2022-01-15:
ah well that's not working. i'm probably just going to give up and add the warning to adblock.

cp2004, 2022-01-15:
What flag are you trying to add?

travisedson, 2022-01-15:
--allow-unauthenticated

cp2004, 2022-01-15:
Is the error the same?

If you can't get past the apt-get update, but you can update the package list manually, then you can skip it if you are editing the script.

Change line 683, either deleting or commenting it:
```diff
-     update_package_list(backup_location)
```

travisedson, 2022-01-15:
yes the error is the same.

doing that allowed the script to run

but now i can't access the pi via ssh and it's at a remote location so i'm dead in the water. oh well i guess.

travisedson, 2022-01-15:
i guess i was being impatient. i am back on the pi but now octoprint doesn't work at all anymore. again, oh well i guess.

cp2004, 2022-01-15:
> again, oh well i guess.

Well would you like more help to try and figure it out or just going to give up straight away?

travisedson, 2022-01-15:
if you're willing to help that's great but it feels pretty hopeless

`pi@octoPi:~ $ sudo apt update
Hit:1 http://download.zerotier.com/debian/buster buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:3 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:5 http://archive.raspberrypi.org/debian buster/main armhf Packages [393 kB]
Get:4 https://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease [1,155 B]
Err:4 https://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease
  The following signatures were invalid: 88E8F32F724468BA39585D4099DA5D2AFCE635A4
Reading package lists... Done
W: GPG error: https://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease: The following signatures were invalid: 88E8F32F724468BA39585D4099DA5D2AFCE635A4
E: The repository 'http://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
pi@octoPi:~ $ sudo apt update--allow-unauthenticated
E: Invalid operation update--allow-unauthenticated
pi@octoPi:~ $ sudo apt update --allow-unauthenticated
Hit:1 http://download.zerotier.com/debian/buster buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Hit:3 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:4 https://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease [1,155 B]
Err:4 https://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease
  The following signatures were invalid: 88E8F32F724468BA39585D4099DA5D2AFCE635A4
Reading package lists... Done
W: GPG error: https://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease: The following signatures were invalid: 88E8F32F724468BA39585D4099DA5D2AFCE635A4
E: The repository 'http://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
pi@octoPi:~ $ sudo apt clean
pi@octoPi:~ $ cd /var/lib/apt
pi@octoPi:/var/lib/apt $ sudo mv lists lists.old
pi@octoPi:/var/lib/apt $ sudo mkdir -p lists/partial
pi@octoPi:/var/lib/apt $ sudo apt update
Get:1 http://download.zerotier.com/debian/buster buster InRelease [36.8 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:5 http://download.zerotier.com/debian/buster buster/main armhf Packages [4,037 B]
Get:6 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 https://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease [1,155 B]
Get:7 http://archive.raspberrypi.org/debian buster/main armhf Packages [393 kB]
Err:4 https://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease
  The following signatures were invalid: 88E8F32F724468BA39585D4099DA5D2AFCE635A4
Get:8 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.8 kB]
Get:9 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Get:10 http://raspbian.raspberrypi.org/raspbian buster/rpi armhf Packages [1,360 B]
Reading package lists... Done
W: GPG error: https://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease: The following signatures were invalid: 88E8F32F724468BA39585D4099DA5D2AFCE635A4
E: The repository 'http://www.linux-projects.org/listing/uv4l_repo/raspbian wheezy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.`

that's where i'm at.

travisedson, 2022-01-15:
i dont know how to insert the code as a block...

travisedson, 2022-01-15:
now trying to add keys and getting more errors

`pi@octoPi:/var/lib/apt $ wget -nv https://download.opensuse.org/repositories/home:strycore/xUbuntu_16.04/Release.key -O Release.key
Release.key: Permission denied`

travisedson, 2022-01-15:
i honestly have no idea what any of this means and i really don't understand why it won't just work, or let me simply bypass the unsigned bit. i don't care if it's signed or not

cp2004, 2022-01-15:
It looks like at some point you tried to add some extra sources to the list, which broke it. I don't recognise the repos that broke as default ones, so I guess it was trying to install something that wasn't in the default repositories as a separate project.

I'll probably need some more details about your OS installed on the Pi because it's not clear what it is? Is is Raspberry Pi OS?

Also, the repositories generating the errors are for raspbian wheezy, when the rest of them are buster, so it was never going to work. If you aren't using Ubuntu 16.04 then this command won't work either:
```
pi@octoPi:/var/lib/apt $ wget -nv https://download.opensuse.org/repositories/home:strycore/xUbuntu_16.04/Release.key -O Release.key Release.key: Permission denied
```
---

I would try the following, to remove the problem repositories:
```
ls /etc/apt/sources.list.d
```
Locate one that looks like `uv4l.list` - it might be called something slightly different. if you can't find one, send the whole list.
Then remove it with `sudo rm /etc/apt/sources.list.d/<the name of the list>`

travisedson, 2022-01-15:
ok,

I deleted one entry which was playonlinux.list

Here is what's left:
`pi@octoPi:/var/lib/apt $ ls /etc/apt/sources.list.d
lutris.list  raspi.list  vscode.list  zerotier.list`

zerotier is my VPN service which I am connecting to the pi through

avahi-daemon running, octopi still inaccessable

travisedson, 2022-01-15:
`pi@octoPi:/var/lib/apt $ neofetch
  `.::///+:/-.        --///+//-:``    pi@octoPi
 `+oooooooooooo:   `+oooooooooooo:    ---------
  /oooo++//ooooo:  ooooo+//+ooooo.    OS: Raspbian GNU/Linux 10 (buster) armv7l
  `+ooooooo:-:oo-  +o+::/ooooooo:     Host: Raspberry Pi 3 Model B Plus Rev 1.3
   `:oooooooo+``    `.oooooooo+-      Kernel: 5.10.63-v7+
     `:++ooo/.        :+ooo+/.`       Uptime: 35 mins
        ...`  `.----.` ``..           Packages: 1425 (dpkg)
     .::::-``:::::::::.`-:::-`        Shell: bash 5.0.3
    -:::-`   .:::::::-`  `-:::-       Terminal: /dev/pts/0
   `::.  `.--.`  `` `.---.``.::`      CPU: BCM2835 (4) @ 1.400GHz
       .::::::::`  -::::::::` `       Memory: 66MiB / 871MiB
 .::` .:::::::::- `::::::::::``::.
-:::` ::::::::::.  ::::::::::.`:::-
::::  -::::::::.   `-::::::::  ::::
-::-   .-:::-.``....``.-::-.   -::-
 .. ``       .::::::::.     `..`..
   -:::-`   -::::::::::`  .:::::`
   :::::::` -::::::::::` :::::::.
   .:::::::  -::::::::. ::::::::
    `-:::::`   ..--.`   ::::::.
      `...`  `...--..`  `...`
            .::::::::::
             `.-::::-`
`

mmasson, 2022-05-18:
I was having the same problem and was able to temporarily disable certificate checking in apt with this config in apt:

```
touch /etc/apt/apt.conf.d/99verify-peer.conf && echo >>/etc/apt/apt.conf.d/99verify-peer.conf "Acquire { https::Verify-Peer false }"
```

You can remove /etc/apt/apt.conf.d/99verify-peer.conf after...

Regards,