cp2004/Octoprint-Upgrade-To-Py3 #66: ERROR: failed to update package list

GreggsNBacon, 2022-10-23:
Hey all!

I have been trying to use this script to update my Octoprint install and I'm getting an error (mentioned in the title.

I'll attach the log from terminal. I don't have any plugins installed (aside from the default ones)
[console_log.txt](https://github.com/cp2004/Octoprint-Upgrade-To-Py3/files/9847607/console_log.txt)

Not sure I understand the script enough to know how to fix this.

cp2004, 2022-10-23:
The script is trying to run `sudo apt-get update --allow-releaseinfo-change` which is supported on all the distributions I was targeting with this script. 

If this is not working for you, then please share details about your install. It sounds like you are trying to update quite an old system so it may not work out for you - I have no idea without more details.

GreggsNBacon, 2022-10-23:
yeah it's an old raspberry pi 3B+ running raspbian.

I'm on Octoprint 1.4 right now. 

Will likely just wipe the SD and install OctoPi instead if that'll work 😄

cp2004, 2022-10-23:
*And what version of Raspbian is that?* Or OctoPi version. Numbers are more helpful - OctoPrint 1.4 doesn't tell me much, it is just an application installed and has no bearing on the OS version.

The recommended route to upgrade to Python 3 has been to flash the latest OctoPi image (0.18), taking a backup of your current OctoPrint install so you can restore your settings with the new version.

Perhaps you should try `sudo apt update` and `sudo apt upgrade` first, but anyway I would recommend the above approach.

GreggsNBacon, 2022-10-23:
No idea anymore! Already wiped it. I know it was installed sometime around 2016-2017 and updated from there, but the install was always troublesome when doing anything. 

Thanks anyways!

closed 2022-10-23 by GreggsNBacon