cp2004/Octoprint-Upgrade-To-Py3 #67: [Bug]

rob3294, 2022-10-26:
**Describe the bug**
It seems that maybe I waited too long to do the update and now the script will not work (see below).  Is there a way to upgrade to an interim version or do I need to start with a fresh install?  If a fresh install, is there a least painful way to do it?

**Script output**
pi@octopi:~ $ curl -L https://get.octoprint.org/py3/upgrade.py --output upgrade.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   154  100   154    0     0    156      0 --:--:-- --:--:-- --:--:--   156
100 27973  100 27973    0     0  20493      0  0:00:01  0:00:01 --:--:-- 20493
pi@octopi:~ $ python3 upgrade.py
OctoPrint Upgrade to Py 3 (v2.2.2)

This script will move your existing OctoPrint configuration from Python 2 to Python 3
This script requires an internet connection and it will disrupt any ongoing print jobs.

It will install the latest version of OctoPrint and all plugins.
No configuration or other files will be overwritten

Press [enter] to continue or ctrl-c to quit

Checking system info...
Detected OctoPi version 0.17.0
Found version: Python 2.7.16
Checking OctoPrint version...
/home/pi/oprint/local/lib/python2.7/site-packages/OpenSSL/crypto.py:12: CryptographyDeprecationWarning: Python 2 is no longer supported by the Python core team. Support for it is now deprecated in cryptography, and will be removed in the next release.
  from cryptography import x509
OctoPrint version: 1.3.12
Please upgrade to an OctoPrint version >= 1.4.0 for Python 3 compatibility

**Script version**
OctoPrint Upgrade to Py 3 (v2.2.2)

**OctoPrint Version**
Version 1.3.12

cp2004, 2022-10-26:
OctoPrint 1.3.12 is the last version that has no Python 3 support, you will need to have a version at least 1.4.0 to use this script. Anywhere from 1.4.x to 1.7.x supports Python 2 *and* 3. You might not be able to do this from the inbuilt updater because it is *so old* it has no concept of the correct versions for Python compatibility like newer installs do. You can still upgrade it manually though.

The process of re-flashing is explained in the main FAQ topic:

https://community.octoprint.org/t/upgrading-your-octoprint-install-to-python-3/35158?u=charlie_powell

cp2004, 2022-10-29:
Since there is nothing I can do about this issue, I'll close it. You can still comment if you have any further follow up questions, but in general if you need further support with OctoPrint the [community forums](https://community.octoprint.org) or [discord server](https://discord.octoprint.org) will be the best places.

closed 2022-10-29 by cp2004