cp2004/Octoprint-Upgrade-To-Py3 #68: Python --version shows 2.7.16 on fresh image.

3DEnthusiast, 2023-01-02:
**Describe the bug**
Trying to install a plugin developed in 3.11, I get an error saying python versions do not match. When checking the python --version it comes back as 2.7.16. 

[image]

**OctoPrint Version**
OctoPrint 1.8.6

jneilliii, 2023-01-02:
Pretty sure OctoPrint doesn't fully support 3.11 right now in the latest stable version of OctoPrint. To get your version of python that octoprint is running on you need to run /home/pi/oprint/bin/python --version, or just look in the web interface at the bottom of the left sidebar.

3DEnthusiast, 2023-01-02:
Thanks for this. Looks like Python 3.7.3. Why would it report 2.7.16?

jneilliii, 2023-01-02:
there are multiple python versions installed on the system, you can have more than one. the one you are seeing when you run python after logging in is considered the default system python and is found in the OS's system path. The one in the other folder I listed is a virtual python environment (virtual environment or venv) that was created using the newer python version as the default interpreter. You probably get the same version information if you were to run python3 --version from direct login (this is the python interpreter used to create the venv).

closed 2023-01-02 by cp2004