cp2004/Octoprint-Upgrade-To-Py3 #69: [Bug]

nablabla, 2023-01-04:
**Describe the bug**
tryping to use option 3: """OctoPi users: [Use this script](https://github.com/cp2004/Octoprint-Upgrade-To-Py3#using-this-script) to upgrade your OctoPi 0.17 install to use Python 3"""
is not working 

**Script output**
[a bunch of numbers which I will not tpye in here]
curl: (6) Could not resolve host: pythong3
curl: (6) Could not resolve host: upgrade.py 

**Script version**
I don't know, i tpyed in this command: curl -L https://get.octoprint.org/py3/upgrade.py --output upgrade.py python3 upgrade.py

**OctoPrint Version**
OctoPrint 1.4.2

**Relevant plugins that may have caused errors**
I don't think any of this is relevant

**Additional information**
sorry, I guess that is all I have

cp2004, 2023-01-04:
There are two commands you have to run, they are on separate lines. It looks like you put them on the same line.

nablabla, 2023-01-04:
dang it..... >.<  kind of feel stupid now. Thanks a lot though!! 👍 👍 👍 👍 👍

closed 2023-01-05 by cp2004