cp2004/Octoprint-Upgrade-To-Py3 #70: [Bug] SSL certificate problem: certificate has expired

kbratos, 2023-02-05:
I get this error:

curl: (60) SSL certificate problem: certificate has expired
More details here: https://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.

but when I use 
  curl -k http://get.octoprint.org/py3/upgrade.py --output upgrade.py
  python3 upgrade.py

I get: 

pi@octopipredator:~ $ curl -k http://get.octoprint.org/py3/upgrade.py --output upgrade.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   154  100   154    0     0    479      0 --:--:-- --:--:-- --:--:--   481
pi@octopipredator:~ $ python3 upgrade.py
  File "upgrade.py", line 1
    <html>
    ^
SyntaxError: invalid syntax

cp2004, 2023-02-05:
I don't see an issue with the SSL cert from here, so I suspect there is something with your network/internet provider that is blocking the connection. Perhaps if you open and read the contents of what was downloaded then it may give clues. The command for curl does include a `-L` in it, in the instructions, which it doesn't look like you've included - that's required to download following the redirect.

kbratos, 2023-02-07:
> I don't see an issue with the SSL cert from here, so I suspect there is something with your network/internet provider that is blocking the connection. Perhaps if you open and read the contents of [...]

pi@octopi:~ $ curl -L https://get.octoprint.org/py3/upgrade.py --output upgrade.py
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (60) SSL certificate problem: certificate has expired
More details here: https://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.
pi@octopi:~ $ python3 upgrade.py

I didn´t paste the first command, I don´t think it´s a network issue because Igot a succesful update 3 weeks ago.

kbratos, 2023-02-07:
curl -k -L  https://get.octoprint.org/py3/upgrade.py --output upgrade.py
python3 upgrade.py

got it to work like this

closed 2023-02-07 by kbratos